PiSupply/PiJuice #900: Powering battery without pi juice hat ?

chrystelenicot, 2022-08-17:
Hello everyone,

a quick question, is it possible to reload power to an external battery (for exemple PIJuice 12000mAh Battery) without the pi juice hat ? 

im thinking to have extra batteries for an interface, but I’m not sure if I can charge a battery without the hat and how.

Thanks a lot in advance !

tvoverbeek, 2022-08-17:
You need a charger for a single cell LiPo battery.
something like this one: https://shop.pimoroni.com/products/adafruit-universal-usb-dc-solar-lithium-ion-polymer-charger-bq24074?variant=32236671828051
Note the charging current and make sure you connect the battery correctly.

chrystelenicot, 2022-08-17:
thanks for your quick answer !
does it exist a charger without the jst connector ? I cut mine... to adapt it to the pi juice hat :"(

tvoverbeek, 2022-08-17:
You probably need to make your own adapter cable.