PiSupply/PiJuice #901: Pi Juice CE compliance

pcou31, 2022-08-23:
Hi, 
I'm looking for the CE certificate of Pi Juice Hat, compliant with rohs amendement 2015/863. 
The certificate on Github (https://github.com/PiSupply/PiJuice/blob/master/Documentation/Compliance/HAT/PiJuice%20HAT%20-%20EC%20Declaration%20of%20Conformity.pdf) indicates only rohs 2011 directive. 
Rohs 2015 is now mandatory for a new product CE marking.
Thanks a lot!
Pierre

tvoverbeek, 2022-08-23:
Note the HAT product is not new. Original design goes back to 2015. The kickstarter version got shipped after very many delays in 2018.
For more info please contact pi-supply directly (sales@pi-supply.com)
For physical mailing address see the pi-supply website (pi-supply.com)

pcou31, 2022-08-23:
Hi, Thanks a lot for your answer. 
Sorry for misspoke, we integrate the pijuice hat in our new product which has to be CE marked, it is the reason we must use rohs 2015 compliant components. Any idea if there is an update CE declaration of conformity?