PiSupply/PiJuice #902: Won't wake

dmkrueger, 2022-09-07:
Hello,

  I have one of your PiJuice Hats on my Raspi 3B+. Running your latest software and firmware, the device will not turn on automatically. For testing I have it set to start every 5 minutes.
Interface shows checkbox "Minutes period" 5 in the field below.

When the system starts up, the Wakeup enabled is never checked, I have the WakeUpEnable script set to run in a crontab job. The event log shows it being called.

 I have never gotten it work, nor has the Set RTC time ever set the time to the RasPi's time.

Any idea's what could cause this?

dmkrueger, 2022-09-08:
That does make it leave the wake enabled checked! Thanks for the tip!
However it will not start up, I did select the "Reset to default config" that didn't reset all the setting like I expected. 
Anything else you can think of? Is there a setting in raspy-config?

dmkrueger, 2022-09-09:
I did set it to start on an hour, and it appeared to start. I guess I must not understand how to configure it to start every X minutes.

tvoverbeek, 2022-09-09:
To select startup every 5 minutes all day and night:
on the Wakeup Alarm Panel (https://github.com/PiSupply/PiJuice/tree/master/Software#wakeup-alarm-menu)
Select Every day, Every hour, Minutes period, Value 5, Seconds 0.
Minutes period 5 it will startup hh:00, hh.05, hh:10, hh:15, ...., hh:55
You have to define something for Day and Hours.
Hope this helps