PiSupply/PiJuice #904: SYS_FUNC_HALT_POW_OFF 60 seconds

zuliani71, 2022-09-16:
Hello, I'm using the pijuce zero with my raspberry pi zero device. Currently, I'm using a headless system to configure pijuce with the pijuice_cli program (not the python version pijuice_cli.py). As I can see from the manual  SYS_FUNC_HALT_POW_OFF still provides power to the Raspberry Pi for a further 60 seconds after shutdown. Is there a way to reduce that value? How?

tvoverbeek, 2022-09-16:
You can modify the value by editing /usr/bin/pijuice_sys.py.
The 60 seconds is defined at line 67 (https://github.com/PiSupply/PiJuice/blob/master/Software/Source/src/pijuice_sys.py)

zuliani71, 2022-09-16:
Ok, thanks I've changed the value inside the python script and it's working now