PiSupply/PiJuice #905: empty json configuration file

zuliani71, 2022-09-16:
Hello, I've configured the pijuice with the pijuice_cli. I was expecting the configuration to be inside the /var/lib/pijuice/pijuice_config.JSON but currently, that file is empty. Is it correct? If yes where is written my current config?

zuliani71, 2022-09-16:
Thanks for the answer but something is not clear to me. When I configure the pijuice with pijuice_cli where the config is written? Inside /var/lib/pijuice/pijuice_config.JSON? Mine  just includes the following string: {"system_task":{"enabled": true}}

tvoverbeek, 2022-09-17:
The JSON file contains only settings which affect the pijuice service (pijuice_sys.py).
Settings which are internal to the PiJuice micro controller (e.g. alarm settings, button settings) are not stored in the JSON file, but in the micro controller non-volatile memory.

zuliani71, 2022-09-20:
Ok, I am asking this question because I would need to configure a large number of pijuce and it would be extremely convenient to be able to do this with an automatic script and not with the interface (for now I use pijuice_cli as my system is headless). What do you think about it? Could there be a solution?

tvoverbeek, 2022-09-20:
Look at https://github.com/PiSupply/PiJuice/tree/master/Software/Source/Utilities.
Especially the --dump and --load options.
This should do (most of) what you want.