PiSupply/PiJuice #907: external led for monitoring system status after button press

zuliani71, 2022-09-16:
Hello, on the pijuce zero I'm using an external button plugged into the J5 terminal. The button includes a led. Currently, that led is plugged into the 5V pin on the pijuce but it will be nice to reproduce on the external led the same behavior of the pijuce status led. Is there a way to implement that?

tvoverbeek, 2022-09-16:
Not easily. you need to solder a connection from the on-board led.
Alternatively you need to write a small monitoring program and connect the external led to an unused GPIO

zuliani71, 2022-09-16:
Ok, about soldering a connection to the onboard led, are there some easy-to-access pads where I can work?

tvoverbeek, 2022-09-17:
Note the PiJuice LEDs are RGB LEDs (3 separate LEDs into one).
See the schematic on github.
I do not have info on where to pick off the signals.
I do not recommend trying it since I do not know the driving power for the on-board LEDs.
Best to implement it with your own code using an available GPIO output.

zuliani71, 2022-09-20:
Ok, thanks for all the suggestions.