PiSupply/PiJuice #908: Pijuice Zero 600mAh battery charing issue (battery temperature wrong)

mroelandts, 2022-09-16:
When connecting a Pijuice Zero 600mAh battery to the Pijuice Zero (firware v1.6)
The status (thorugh pijuice_cli) reports that the battery is -8 Celsius.
Which is obviously not the case!
But therefore the battery is not charging:
`Fault: charging temperature SUSPEND`
When I hold the battery in my hand, I get the temperature to rise to -4 Celsius.
Which means the sensor is working but just got a terrible offset??
If I unplug the battery and after a minute plug it back in, the first five seconds are normal (24 degrees) and start charging but after that back to -x degrees.

If I go to the battery profile, I see that the wrong battery is automatically detected.
It reports a PJZERO_1000 is connected. Changing this to a PJLIPO_600 does not fix anything...
The only way to make it charge, is by changing the "temperature sense" to "ON_BOARD" or "NOT_USED".
Since my raspberry (zero) will somethings be place in a box in the sun, I really need the temp of the battery...

I have 2x 600 battery and 2x pijuice zero boards. Any combination has the same issue!
Is this a firmware issue or do I have 2 faulty batteries or 2 faulty zero boards?

tvoverbeek, 2022-09-17:
What is your 'Select Rsoc estimation' setting?
For the PiJuice Zero it should be 'Direct by MCU' since the PiJuice Zero does not have a the fuel gauge chip.

mroelandts, 2022-09-22:
The Rsoc estimation was set to "AUTO_DETECT".
Since there is only 1 other option to chose from (DIRECT_BY_MCU), I would guess it always this setting on a zero?
Either way manually changing this to the DIRECT_BY_MCU settings did not help.
Everything is exactly the same.