PiSupply/PiJuice #911: PiJuice sensor Documentation

IanAdd, 2022-10-08:
I'm trying to get a full grasp of what PiJuice sensors tell me. I have one installed and working on a Pi4 running Home Assistant. There is an add-on which exposes a range of information into HA but I don't understand them fully.

As a hopefully helpful start here is a list of the sensors I see and some of the values seen. Has anyone got a full spec?

**PiJuice Battery Status**
States:

- normal - ???
- charging from 5v io - The PiJuice Battery is not fully charged and is being topped up via the GPIO bus.
- ?? other states ?
- 

**PiJuice Power Input status**
States:

- Weak - ???
- Not present - ???
- ?? other states ?

**PiJuice Power input IO status**
States:

- Present - connected to a power source ??
- Bad - ??
- ?? other states ?
- 

**PiJuice Charge**
State:

- example 91% - how full the PiJuice Battery is

**PiJuice Temperature**
State:

- example 24c - ? Battery / Board Temperature ? What temp should be of concern?
- example 9c - how can this temp be correct when the ambience temp is over 20c ?

**PiJuice Battery Voltage**
State:

- example 4.158V - ?? what value range is normal and what should be of concern ?

**PiJuice Battery current**
State:

- example 0.0 A - the current entering/charging the PiJuice from its own micro USB power source ??

**PiJuice IO voltage**
State:

- example 5.18v - the voltage the the PiJuice is delivering to the Raspberry Pi
- NB This value should be in the range ?? to ??

**PiJuice IO current**
State:

- example 0.284 A - the current being drawn from the PiJuice by the Raspberry Pi
- NB if this value is negative then the PiJuice is presumably being charged from the Raspberry Pi GPIO bus, PiJuice Battery Status will be 'charging from 5v io'

tvoverbeek, 2022-10-08:
For more details see the Software Readme on github and the source code of pijuice.py

IanAdd, 2022-10-09:
@tvoverbeek , thanks so much. That was a strange place (in my view) to find end-user documentation but it was what I needed. I will remember to look in those sort of places next time.

closed 2022-10-09 by IanAdd