PiSupply/PiJuice #912: After firmware upgrade: is reboot or wait needed?

OhSoGood, 2022-10-09:
Hi,

After an firmware upgrade of the pijuice, are we supposed to reboot the raspberry and/or the pijuice? Or to wait for a while, notably if we want to set its parameter (via the API, not via pijuice_cli) ? I've found nothing in the documentation about that.

Thank you for your tip

tvoverbeek, 2022-10-09:
If you updated via the GUI or CLI you see a message at the end 'Waiting for restart'. You see it also on the PiJuice LEDs blinking. No Pi reboot needed.
Both thr CLI and GUI use pijuiceboot to do the update. Only reason to use pijuiceboot directly is if you want to downgrade the firmware to an earlier version.

OhSoGood, 2022-10-09:
I can see the code in pijuice_cli.

Can pijuice_cli be used within a shell script to automate the firmware 
version check and upgrade?

I would need that to automate the installation (I have many pijuice to 
setup).
I can of course duplicate and adapt the code of the class FirmwareTab 
from pijuice, but code duplication is never good.

tvoverbeek, 2022-10-09:
Cherry  pick the code in the update firmware tab for your automatic installation.
You cannot use pijuice_cli for automatic updates.
Follow the same logic as in the CLI/GUI:
- check current firmware version
- compare with new firmware version
- if sufficient power available start the update using pijuiceboot if there is a newer version
- check the return code from pijuiceboot
- if OK, try to read pijuice status wit a timeput. If successful firmware has restarted.