PiSupply/PiJuice #915: Installing pijuice python package

mahotd, 2022-10-18:
Hello,

I've installed pijuice-base (`sudo apt-get install pijuice-base`) on my Raspberry.
However, it has installed the pijuice-base package in my system's Python. I would like to be able to use the `pijuice` (i.e. `pip import pijuice`) python module in a virtual env.
Is the `pijuice` package pip-installable ? Is there any way to cleanly make it available in any virtualenv ?

Thanks

tvoverbeek, 2022-10-18:
As in #571 use the `--system-site-packages` switch when creating the virtual environment.
See https://docs.python.org/3/library/venv.html#creating-virtual-environments

mahotd, 2022-10-20:
Thank you for your answer. However, I'm using pyenv-virtualenv and it seems that providing the `--system-site-packages` option doesn't work, even though it should as pyenv supports virtualenv's options...
At the moment, I just `cp` the file to my directory. I'll try in the future to either use virtualenv directly or see how to fix the `--system-site-packages` with pyenv.