PiSupply/PiJuice #921: Pi never completey shutsdown

rowland-cook, 2022-10-23:
I have a Raspberry PI with a USB Sata SSD and is powered over POE.

I have tried SYS_FUNC_HALT_POW_OFF, SYS_FUNC_SYS_OFF_HALT and the shutdown.py.
In all cases when I remove the POE network cable I get the same result. The OS appears to shutdown cleanly, but power still remains on the Pi. A light on the USB to SATA board remain on.
Plugging in the POE network cable does NOT restart the Pi.
I have to remove and replace the battery in order to get the Pi to start.

What is wrong

tvoverbeek, 2022-10-23:
Need more details. How do you shutdown the Pi when the POE cable is removed? 'No Power' event?
System Task settings?
It seems the Pi still gets 5V (from the PiJuice ?) after the POE cable is removed.

rowland-cook, 2022-10-23:
Sorry, yes it is the system events -> no power event.

tvoverbeek, 2022-10-24:
With SYS_FUNC_HALT_POW_OFF the PiJuice removes the 5V to the GPIO 30 seconds after shutdown starts.
If the power led of the Pi continues to be lit it must still get 5V power from somewhere else. Maybe via the USB SATA SSD?
Does the SSD have its own power supply?

rowland-cook, 2022-10-24:
No, there is no power to the USB SATA SSD. the only power is through the
POE cable

rowland-cook, 2022-10-24:
Problem solved.

I was plugging in the POE cable as soon as the PI shut down, which seemed
to be 10 seconds. Had I waited longer (30 seconds) then the PI would have
shutdown.

Sorry just did not realise it took 30 seconds for it to shutdown as opposed
to 10 seconds

Sorry an inconvenience

Many thanks

Rowland Cook

tvoverbeek, 2022-10-24:
Glad you found the solution.
Closing the issue

closed 2022-10-24 by tvoverbeek