PiSupply/PiJuice #922: JSON config file in restricted folder

mahotd, 2022-10-24:
Hi,

When I want to `cd` into `/var/lib/pijuice` to edit `pijuice_config.JSON`, I get the following message:
```bash
pi@raspberry:/var/lib $ cd pijuice/
-bash: cd: pijuice/: Permission denied
```
Indeed, running `cd /var/lib/ && ls -al` returns:
```bash
...
drwxr-xr-x  2 root    root    4096 Sep 22 02:06 pam
drwx------  2 pijuice pijuice 4096 Oct 19 10:55 pijuice
drwxr-xr-x  2 root    root    4096 Sep 22 02:34 plymouth
...
```

Why is the `pijuice` folder restricted and should I make itself and its content accessible to my user (`pi`) ? And if so, what would be the command, something like `chmod 755 /var/lib/pijuice` maybe ?

Thanks !

tvoverbeek, 2022-10-24:
The pijuice folder and JSON file are owned by user pijuice.
You can do `sudo -u pijuice ls -l /var/lib/pijuice` or make it available for all users: `sudo chmod -R 755 /var/lib/pijuice`.

mahotd, 2022-10-24:
Thanks for your quick answer. What is the rationale behind making it owned by the `pijuice` user ? Because the `pijuice` user runs all the pijuice scripts ?