PiSupply/PiJuice #926: Problem with battery temperature reporting?

nikyboydd, 2022-11-04:
The battery temperature reporting seems to be wrong on one of my PiJuice units (or perhaps all of them).  It's reporting that the battery is 47C, however I find that hard to believe as it doesn't seem hotter to the touch than two other identical PiJuice units in identical set ups (and whose battery temp reports 24C).  Also, as you can see on the following graph, the temperature often appears to oscillate suddenly between 24C and 47C, which I find even harder to believe!

[image]

This is causing a problem for me because it means that the battery's temperature often exceeds the warm temperature threshold of 45C, and then the PiJuice refuses to charge it.  The result is that the battery level falls over time, which is obviously not ideal when I'm trying to use it as a UPS.

I guess I could increase the warm battery temp threshold slightly with a custom battery profile to prevent this, but I'm not sure what the consequences of doing that would be, ie might it damage the battery (a BP7X)?

nikyboydd, 2022-11-04:
Funnily enough one of my other units has just started exhibiting similar behaviour today - the battery temperature, which previously had been hovering steadily around 24C for days, suddenly started oscillating wildly between 24C/25C and 46C for ten minutes or so before finally settling at 46C, where it's remained since:

[image]

Admittedly I have been stressing this particular unit a bit, which no doubt pushed the overall temperature up a bit, but I find it hard to believe that it would do so by 20C?  It seems like the battery temp was in fact 26C/27C rather than 46C/47C, ie it almost seems like the battery temp reporting is erroneously adding 20C to the real value once it goes above 25C?

Has anyone else seen this behaviour?  All my units are running firmware 1.6.

nikyboydd, 2022-11-04:
I've just noticed in issue #720 that someone else saw the same thing, where their battery temperature reading suddenly dropped from 47C down to 24C (see https://github.com/PiSupply/PiJuice/issues/720#issuecomment-833847161) which mirrors what I am experiencing?

nikyboydd, 2022-11-04:
After further searching on here I've found more issues that mention the same problem with the temperature reading so it seems like it is a common thing, although I've not been able to find yet any mention of a real solution (apart from disabling the temperature sensor completely, which doesn't seem like a great fix to me!)

For now, I've adjusted the battery profile by increasing the warm temperature threshold to 49C so that at least the PiJuice still charges the battery when the readings are wong like this, and have updated my monitoring scripts to adjust the temperature readings to what I think are more correct values, as I'd still like to keep any eye on the temperatures...

maxsitt, 2022-11-07:
> Funnily enough one of my other units has just started exhibiting similar behaviour today - the battery temperature, which previously had been hovering steadily around 24C for days, suddenly started oscillating wildly between 24C/25C and 46C for ten minutes or so before finally settling at 46C, where it's remained since:
> 
> [image]
> 
> Admittedly I have been stressing this particular unit a bit today, which no doubt pushed the overall temperature up a bit, but I find it hard to believe that it would do so by 20C? It seems like the battery temp was in fact 26C/27C rather than 46C/47C, ie it almost seems like the battery temp reporting is erroneously adding 20C to the real value once it goes above 25C?
> 
> Has anyone else seen this behaviour? All my units are running firmware 1.6.

I've been using several PiJuice Zero boards with different 12,000 mAh PiJuice LiPo batteries and also seen this behaviour of oscillating temperatures, independent of the respective hardware (current firmware and software installed). I also searched for a solution for this problem, but found no clear answer what could be the cause. I would assume that it is either a software/firmware problem or some of the PiJuice batteries have faulty NTC sensors, as there are several other open and closed issues regarding this problem (e.g. https://github.com/PiSupply/PiJuice/issues/659#issuecomment-860545300).

Some info on the battery temperature calculation can be found here: https://github.com/PiSupply/PiJuice/issues/891

Possible workaround: https://github.com/PiSupply/PiJuice/issues/625#issuecomment-761884691

tvoverbeek, 2022-11-07:
There is/was a batch of 12000 mAh batteries with different values for the NTC parameters.
See https://github.com/PiSupply/PiJuice/blob/master/Software/Configs/Batteries/LIPO_12000_PIS-1129.JSON
Note the difference for the NTC parameters:
Firmware: ntcB = 3450, ntcResistance = 10000
Alternate: ntcB = 5200, ntcResistance = 42000
Try with the alternate profile by defining it as custom profile.and see it this improves the situation.