PiSupply/PiJuice #928: How to shut down raspberry pi at specifict time with Pi Juice

jorgrde1, 2022-11-08:
Hi, I am starting on a Raspberry pi 4B with Pi Juice hat and I'm trying to shutdown it at a specific hour, I'm only finding scripts which did it but only woks with minutes like, shutdown after 30,120,150 mins. It is possible to do it with an specific hour and how?

jorgrde1, 2022-11-08:
I've done by writing ```/sbin/shutdown -h 10:00``` before the exit 0 line on rc.local by ```sudo nano /etc/rc.local```

But the red LED on Raspberry Pi still on and didn't turn off just like when I do the SYS_FUNC_HALT_POW_OFF

tvoverbeek, 2022-11-08:
If you want to shut down at a specific time / weekday/ date use a cronjob to schedule the shutdown.
See 'man cron' and 'man crontab' for details.
On the PiJuice make sure you have enabled 'Software Halt Power Off' to remove the 5V power to the Pi after the software 
 initiated shutdown.
The PiJuice cannot schedule a shutdown at a specific time, only a wakeup at a specific time.

jorgrde1, 2022-11-09:
Done. Thank you so much!

tvoverbeek, 2022-11-09:
Closing the issue

closed 2022-11-09 by tvoverbeek