PiSupply/PiJuice #930: PiJuice Battery and GPIO Input Issues

EzequielRosario, 2022-11-20:
Good day. I'm working with a PiJuice HAT for a few months and it's been working great! However, since last week I've had problems with it since it doesn't power on by itself when the battery is connected nor when it's USB Input is connected either. The only way I was able to make it work is by connecting the USB Input to my Raspberry Pi 4B model, which is what Im working with. I can disconnect it afterwards and it keeps running with the battery and the USB Input from the PiJuice if connected just fine, but once I turn it off, it won't turn on with the PiJuice in any way, nor the Pijuice will show the blinking green/red light showing that it's in deep sleep mode. The battery is a PJLIPO12000mAh model, and the PiJuice Firmware is 1.6

Performing some debugging on the PiJuice GUI I get these results:
(With both RPi and Pijuice cables connected)
[image]

(With only Pijuice cable connected)
[image]

(With neither cable connected)
[image]

(Current settings)
[image]
[image]
[image]
[image]
[image]

Additional notes: Sometimes GPIO Input changes to Present, Bad or Weak, but mostly stays between Weak and Bad. I've tried multiple external power sources (direct connection to wall, power bank, different PiJuice battery, etc.), but no changes.
Any responses soon would be greatly appreciated, as I am in a bit of a hurry for a college project haha.
If possible to have additional tech support, I'm on the Pi Supply Discord as ZeeK.

tvoverbeek, 2022-11-20:
Does the Pi turn on on button (SW1) press?
Does  the Pi have any peripherals connected?
It smells like the load of the Pi + peripherals is too large for the power supply (no reserve to charge the PiJuice).
Even with power connected to the Pi only the 'GPIO Power Input' shows weak and the voltage (4.94) seems low.

EzequielRosario, 2022-11-20:
> Does the Pi turn on on button (SW1) press? Does the Pi have any peripherals connected? It smells like the load of the Pi + peripherals is too large for the power supply (no reserve to charge the [...]

It does not turn on with SW1 pressed.
Currently it has a camera module connected, and the GPIO is connected by a ribbon with a motion sensor, a temperature/humidity sensor and a small fan for the CPU.
It could be a load problem, but the issue came up so suddenly, without any change in the hardware. I will disconnect everything and try it out.
Also, just for clarification, the PiJuice is completely shut down, even with its battery and USB input connected, regardless of being connected to the Pi's GPIO.

EzequielRosario, 2022-11-20:
Well, now this is weird...
Using the hardware setup below, I receive >5V on GPIO Input but still shows a WEAK status, with no additional load on it.
[image]
[image]

tvoverbeek, 2022-11-20:
The issue could be your 40 wire flat cable which has to power the 4B.
Can you connect the PiJuice directly (without the heat sinks)
Alternatively you can use an 40 pin extension header to raise the PiJuice so it does not interfere with the heat sinks.
By the way why do you need heat sinks? Overclocking? Heavy compute load?

EzequielRosario, 2022-11-20:
Thanks for the responses, by the way.
Can the heat sinks be part of the problem? Cuz if not I can place the pijuice directly without them interfering.
I need the heat sinks because my project involves placing this system in a small enclosed space while using solar power to charge it over time. I've also made scripts to automatically start it up should it shut down, and they've worked so far.

tvoverbeek, 2022-11-20:
WIth the heatsinks in place is there sufficient room to put the PiJuice directly on the Pi's 40-pin GPIO header?
What I meant the heatsinks themselves are not the problem, but connecting the PiJuice to the Pi via the flat cable.

EzequielRosario, 2022-11-20:
> 

At the moment the PiJuice is directly connected to the RPi's 40-pin GPIO header, and has the results shown above.

tvoverbeek, 2022-11-20:
OK. THen the PiJuice believes your power supply is too weak.
Can you try to reflash the firmware and see it the situation improves?

EzequielRosario, 2022-11-20:
I am currently reinstalling the Pijuice firmware. I will update the post with the results.
Is it necessary to reflash my entire SD card for this, in case this doesn't work?

tvoverbeek, 2022-11-20:
No, no need to reflash the complete SD card.

EzequielRosario, 2022-11-20:
I uninstalled PiJuice using the commands in terminal:
`sudo apt-get remove -y pijuice-base
sudo apt-get remove -y pijuice-gui`
I rebooted then reinstalled with `sudo apt-get install pijuice-gui` but no change.

tvoverbeek, 2022-11-20:
I meant the PiJuice firmware, not the Pijuice packages.
In a terminal go to `/usr/share/pijuice/data/firmware/` and 
run `pijuiceboot 14 PiJuice-V1.6_2021_09_10.elf.binary`.
See also https://github.com/PiSupply/PiJuice/blob/master/Firmware/README.md

EzequielRosario, 2022-11-20:
Im getting the following error:
[image]

tvoverbeek, 2022-11-20:
You forgot the address (14) in `pijuiceboot 14 PiJuice-V1.6_2021_09_10.elf.binary`.
sudo is not needed

EzequielRosario, 2022-11-20:
Whoops, my bad.
Well I did it and it executed. I then assigned the battery to the LIPO12000, but sadly the same results show.
[image]

EzequielRosario, 2022-11-20:
I believe the problem is in the PiJuice hardware itself not drawing from its sources, as the PiJuice is not powered even while not having connection to the RPi's GPIO. Pressing SW1 in this state does not turn on the LED.

EzequielRosario, 2022-11-21:
UPDATE: I did an additional test using a power bank and connecting it to the USB input of the Pijuice to power it up with the same setup as the one above. It turns out the power bank is not giving any output while the Pijuice is off, not even when I press the SW1 button. 
Hence, the PiJuice is not automatically receiving/giving power whatsoever unless it's connected to a RPi that's turned on.

tvoverbeek, 2022-11-21:
Powerbanks typically require a minimum load, otherwise they switch off.
Obviously the PiJuice does not draw enough current from the powerbank.

When you power via the Pi is the Pijuice led blinking (blue/red when battery < 50%, blue/green when battery > 50%)?
Your screenshots show the battery is charging when powering via the Pi.
Then when you disconnect the Pi power does it continue running, or does it shut down abruptly, or orderly after some time.

It looks like there is a problem with the PiJuice power input via USB.
Do you have the possibility to connect your external power to J4?
From the hardware README:
J4 is an alternative to the PiJuice microUSB micro input. Pad holes are provided to solder a 2.54mm header...any type (straight or right angle, female or male) and length can fit. You can plug in power sources to this connector with input voltage between 4.2V and 10V. A minimum current of 80mA will be sufficient for the battery to charge. Note: Although this connector may be used for some types of solar panels be aware that we only support the use of the official panels.

EzequielRosario, 2022-11-21:
When powered, the PiJuice works as intended for everything related to it except charging at its usual speed. The rest works as intended (GPIO functions, LED blinking while charging, even disconnecting the RPi input cable to leave it on battery power)
I did notice that when I press the SW1 while the RPi is on and running on battery, the RPi and the PiJuice shut down immediately. The single press is assigned to HARD_FUNC_POWER_ON.
I unfortunately don't have a J4 connector for it, but then again I don't have prior knowledge of what it is, but I'll take a look.

tvoverbeek, 2022-11-21:
For J4 look on the hardware page: https://github.com/PiSupply/PiJuice/tree/master/Hardware

[image]

EzequielRosario, 2022-11-21:
It seems i will need additional hardware for it. I can look into shops nearby to test it out.
However, I'm having my doubts that the USB Input is the problem, since the GPIO Input is showing as WEAK regardless of the Micro USB Input.
[image]
[image]