PiSupply/PiJuice #932: Ubuntu 22.04 Installation Error - dh: error: Compatibility levels before 7 are no longer supported (level 1 requested)

WilliamB2022, 2022-11-26:
Evening, 

I have been trying for the past few days to install a pijuice on an Ubuntu 22.04.1 LTS 64bit.
I have tried various attempts and searched through the issues and commits by @tvoverbeek to no avail.

I am following line for line the guide [here](https://github.com/PiSupply/PiJuice/blob/master/Software/OS%20Support/ubuntu_server_64.MD) and always getting the same error when it comes to running `sudo ./pckg-pijuice.sh`

Please see below error code...
` dpkg-source --before-build .
 fakeroot debian/rules clean                                                                                                   
dh clean --with python3 --with systemd --buildsystem=pybuild                                                                   
dh: error: Compatibility levels before 7 are no longer supported (level 1 requested)
make: *** [debian/rules:5: clean] Error 255
dpkg-buildpackage: error: fakeroot debian/rules clean subprocess returned exit status 2
running bdist_deb
`

I have attempted various ways to rectify the error. Even altering the dh-compat version within stdeb.cfg.
For some reason it seems the install just cant seem to see the specified compat version and instead is giving it version 1.

has anyone come across this error or knows of a solution please?

WilliamB2022, 2022-11-27:
I believe I have managed to get past the above issue. 
The control file needed debhelper-compat (= 10) instead of debhelper (>= 10)

I have now ran into another issue of `pybuild pybuild:127: cannot detect build system, please use --system option of set PYBUILD_SYSTEM env. variable.`

cc68, 2022-12-04:
I although have some difficulties to get the pijuice software properly running ob Ubuntu 22.04lts and rpi4.
When following the installation instructions you mentioned, you have to replace 
sudo apt install ./pijuice-base_1.6_all.deb
with
sudo apt install ./pijuice-base_1.8_all.deb
and do the same vor the GUI Version.
Does that help?

tvoverbeek, 2022-12-04:
@cc68 I just updated the instructions in the OS Support section to use version 1.8.

WilliamB2022, 2022-12-07:
> I although have some difficulties to get the pijuice software properly running ob Ubuntu 22.04lts and rpi4. When following the installation instructions you mentioned, you have to replace sudo apt [...]

Thank you for your reply, I was already building with 1.8 as I noticed the update was available.
After figuring out the compat change on the control file I entered into a deep deep rabbit hole of problems that I was chipping away at for several days. I ended up at a state where I could load the CLI UI however it seemed that I was unable to alter any settings.

I have taken a pause on attempting to hack away at trying to make it work as the problem was taking up far too much of what should of been a quick fix. 

I will come back to after Christmas and take another shot. If in the meantime someone does happen to find a way to get pijuice working on Ubuntu 22, I will be very keen to see how you managed it.