PiSupply/PiJuice #933: Temperature at 24º C but its really at 40ºC

jorgrde1, 2022-12-01:
Hi, I working on a Raspberry Pi 4B with a PiJuice hat on it for a month and it from one day to another the battery level get fixed at 86% and the temperature is at 24º C.
I have seen in another issues that the battery problem may be solved by time, but I dont know why the temperature is fixed on 24, I used a thermometer to measure the temperature and it marks +40.
Software version: 1.8
Firmware version: 1.5
OS version: Linux 5.15.76-v7

tvoverbeek, 2022-12-01:
Have you tried to update the firmware to 1.6 and see if the problem persists?

jorgrde1, 2022-12-02:
Where can I find this firmware version?
I have all my raspberry software up to date already, I didn't find any version of the firmware at 1.6.

jorgrde1, 2022-12-02:
During the morning today it seems that it has fixed itself, I have not done anything to the software or hardware, it has simply started to show the temperature data well and the battery has charged up to 99% like the first day I used it.
[image]

jorgrde1, 2022-12-02:
I think it could be a software bug.

tvoverbeek, 2022-12-02:
From your screenshot I see you are not using the PiJuice software utilities like pijuice_gui or pijuice_cli.
The 1.6 firmware is part of the 1.8 pijuice-base package.
If in the pijuice_gui or pijuice_cli you go to the firmware tab it should show a newer version is available.
Anyway it should be located in /usr/share/pijuice/data/firmware/[PiJuice-V1.6_2021_09_10.elf.binary
See also https://github.com/PiSupply/PiJuice/tree/master/Software/Source/data/firmware

jorgrde1, 2022-12-09:
Yes, I am using the pijuice_gui, my screenshot its only a Thingsboard panel where I plot the pijuice data send by mqtt.
I have already installed the 1.8 pijuice-base package and reinstalled it, doesnt appear the 1.6 firmware.

tvoverbeek, 2022-12-09:
Something is wrong. If you installed pijuice-base from the Raspberry Pi repos the 1.6 firmware should be there.
Here is my output of the firmware tab in pijuice_cli:
[image]
If the installed firmware is older it should prompt you to update the firmware.

Also can you run in a terminal `dpkg -L pijuice-base`.
Here is the expected output:
```
tvo@rpi32-deb11:~ $ dpkg -L pijuice-base
/.
/etc
/etc/sudoers.d
/etc/sudoers.d/020_pijuice-nopasswd
/etc/udev
/etc/udev/rules.d
/etc/udev/rules.d/99-i2c.rules
/lib
/lib/systemd
/lib/systemd/system
/lib/systemd/system/pijuice.service
/usr
/usr/bin
/usr/bin/pijuice_cli.py
/usr/bin/pijuice_cli32
/usr/bin/pijuice_cli64
/usr/bin/pijuice_log.py
/usr/bin/pijuice_sys.py
/usr/bin/pijuiceboot32
/usr/bin/pijuiceboot64
/usr/lib
/usr/lib/python3.9
/usr/lib/python3.9/dist-packages
/usr/lib/python3.9/dist-packages/__pycache__
/usr/lib/python3.9/dist-packages/__pycache__/pijuice.cpython-39.pyc
/usr/lib/python3.9/dist-packages/pijuice.py
/usr/lib/python3.9/dist-packages/pijuice_base-1.8.egg-info
/usr/share
/usr/share/doc
/usr/share/doc/pijuice-base
/usr/share/doc/pijuice-base/changelog.gz
/usr/share/pijuice
/usr/share/pijuice/data
/usr/share/pijuice/data/firmware
/usr/share/pijuice/data/firmware/PiJuice-V1.6_2021_09_10.elf.binary    <====
tvo@rpi32-deb11:~
```