PiSupply/PiJuice #934: PiJuice hat switches

johanjohan, 2022-12-11:
I connected the micro USB to the PiJuice directly.
I am having issues pressing the switches. SW1 sticks "inside" once pressed (and the others behave similar), so I cannot easily control power on and off like that. Is this the normal behaviour?
Is there a tutorial video online? I have already watched a dozen YouTubes without enlightenment. I would like to know whether the switch is defunct or I am having false assumptions concerning operation.

tvoverbeek, 2022-12-11:
SW1 (and the other switches) are momentary push. They should not stick. Definitely a defective switch.
Note: You can use an external pushbutton  via J5 which is parallel ti SW1.
See https://github.com/PiSupply/PiJuice/tree/master/Hardware#switches

johanjohan, 2022-12-11:
@tvoverbeek thanks, that is what i thought. is it reasonable to apply some WD40 or is that a bad idea? the PiJuice is brandnew, shall I rather contact the vendor?

tvoverbeek, 2022-12-11:
I suggest you contact the vendor and try to get a replacement

closed 2022-12-14 by johanjohan