PiSupply/PiJuice #936: SYS_FUNC_HALT_POW_OFF does it measure current?

GekoCH, 2022-12-21:
Hy

I have a Pi 3 running PiJuice with an NFC reader attachted to I2C and the 5v port of the Rpi.
When I power down the pi with 
```
from pijuice import PiJuice

import os
pijuice = PiJuice(1, 0x14)

# Remove 5V power to RPi after 30 seconds
pijuice.power.SetPowerOff(30)

# Shut down the RPi
os.system("sudo poweroff")
```
it won't turn off the 5v power rail after 30s. The RED LED on the pi is still on and on the NFC reader as well....
Does the pijuice measure current and if the current is low enough it will switch then 5v rail finally off?

pijuice is running the latest firmware (1.6)

GekoCH, 2022-12-22:
ok I conducted some tests with a "new" pijuice which is working fine. same firmware same config but one is working (switch all 5v devices off) and one is not switch the 5v rail off...
Can someone tell me what may cause this and if a component on the pijuice is defective?
Or is it a bad programmed EEPROM?

Or any other suggestions?

tvoverbeek, 2022-12-22:
For SYS_FUNC_HALT_POW_OFF to work, the PiJuice service has to be running. Did you check on the non-functioning Pi that the service is running?

GekoCH, 2022-12-22:
> For SYS_FUNC_HALT_POW_OFF to work, the PiJuice service has to be running. Did you check on the non-functioning Pi that the service is running?

The service is running:

pi@raspberrypi:~ $ sudo systemctl status pijuice.service
● pijuice.service - PiJuice status service
     Loaded: loaded (/lib/systemd/system/pijuice.service; e>     Active: active (running) since Thu 2022-12-22 11:17:16>   Main PID: 373 (pijuice_sys.py)
      Tasks: 1 (limit: 1597)
        CPU: 2.019s
     CGroup: /system.slice/pijuice.service
             └─373 /usr/bin/python3 /usr/bin/pijuice_sys.py

Dec 22 11:17:16 raspberrypi systemd[1]: Started PiJuice started

tvoverbeek, 2022-12-22:
Only power source is the PiJuice, so the Pi is running on battery. Correct?
How is the NFC reader powered? What do you mean by 'connected to the 5v port of the RPi'? One of the USB ports?

GekoCH, 2022-12-22:
> Only power source is the PiJuice, so the Pi is running on battery. Correct? How is the NFC reader powered? What do you mean by 'connected to the 5v port of the RPi'? One of the USB ports?

I no run the Pi only with the pijuice nothing else attached and it is still not powering down.
The only time I see it sleeping is when I disconnect the battery and plug it in. Then I can see the Green/Red led of the pijuice lit up for a few seconds and then the pijuice starts blinking so it is in sleep.
But once I start the pi by using SW1 on the pijuice I can't get it back to sleep after shutdown...

tvoverbeek, 2022-12-22:
Try the following:
In Pijuice GUI or CLI go to the System Task tab.
Select System Task enabled
Select Software Halt Power Off with a delay period of your choice (e.g. 30).
[image]

In a terminal window (either on Desktop or via ssh or your own script) shut down the system (`sudo poweroff`)
This shuts down the system and turns off the 5V after the specified delay.
Does this work?

GekoCH, 2022-12-22:
thx for the help but no the green/red led of the pijuice is still on after 2 minutes after shutdown...

GekoCH, 2022-12-22:
I think it is a hardware problem or the uC on the pijuice is not working correctly...
Is the 5V rail switched by a transistor? But why can it be powered up correctly with the pushbutton since when I push the button 5v is feeded to the pi. Shutting down would just do the opposite

tvoverbeek, 2022-12-22:
Then I am lost.
Have you tried reflashing the 1.6 firmware on the not working PiJuice?

GekoCH, 2022-12-22:
Jep done that...
Is there a factory reset function?

tvoverbeek, 2022-12-22:
Have you tried the button shutdown? Long Press1 for SW1 with function SYS_FUNC_HALT_POW_OFF. Parameter is press time duration in msec.

GekoCH, 2022-12-22:
Jep done that too but same result. I tried the custome script for a shutdown as well but no success.
Do you know what kind of components might cause this problem on the pijuice? Maybe a trani is malfunctioning..

tvoverbeek, 2022-12-22:
Look at the schematic. Probably one of the voltage converters.
See https://github.com/PiSupply/PiJuice/blob/master/Documentation/PiJuice_Schematic.PDF.
Also the firmware source is on github.