PiSupply/PiJuice #937: Unable to connect to device with Rpi Zero WH on latest DietPI

cserl, 2022-12-30:
5000mAh battery is connected. The two color Led is blinking Green/Blue.

dietpi-config advanced options:
I2C State: On
I2C frequency 100 kHz
Spi state: On

1. start pijuice_cli 
2. select the "General" menu entry 
3. Error message: "Unable to connect to device"

root@DietPi:~# i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

Running pijuice_cli with status:

root@DietPi:~# pijuice_cli

Traceback (most recent call last):
  File "/usr/bin/pijuice_cli.py", line 2087, in <module>
    loop.run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 287, in run
    self._run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 385, in _run
    self.event_loop.run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 790, in run
    self._loop()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 827, in _loop
    self._watch_files[fd]()
  File "/usr/lib/python3/dist-packages/urwid/raw_display.py", line 416, in <lambda>
    wrapper = lambda: self.parse_input(
  File "/usr/lib/python3/dist-packages/urwid/raw_display.py", line 515, in parse_input
    callback(processed, processed_codes)
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 412, in _update
    self.process_input(keys)
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 513, in process_input
    k = self._topmost_widget.keypress(self.screen_size, k)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 598, in keypress
    return self.top_w.keypress(self.top_w_size(size,
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 1626, in keypress
    key = self.focus.keypress(tsize, key)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 2316, in keypress
    key = w.keypress((mc,) + size[1:], key)
  File "/usr/lib/python3/dist-packages/urwid/decoration.py", line 622, in keypress
    return self._original_widget.keypress(maxvals, key)
  File "/usr/lib/python3/dist-packages/urwid/listbox.py", line 968, in keypress
    key = focus_widget.keypress((maxcol,),key)
  File "/usr/lib/python3/dist-packages/urwid/decoration.py", line 622, in keypress
    return self._original_widget.keypress(maxvals, key)
  File "/usr/lib/python3/dist-packages/urwid/wimp.py", line 543, in keypress
    self._emit('click')
  File "/usr/lib/python3/dist-packages/urwid/widget.py", line 461, in _emit
    signals.emit_signal(self, name, self, *args)
  File "/usr/lib/python3/dist-packages/urwid/signals.py", line 265, in emit
    result |= self._call_callback(callback, user_arg, user_args, args)
  File "/usr/lib/python3/dist-packages/urwid/signals.py", line 295, in _call_callback
    return bool(callback(*args_to_pass))
  File "/usr/bin/pijuice_cli.py", line 1994, in item_chosen
    callback()
  File "/usr/bin/pijuice_cli.py", line 209, in __init__
    self.main()
  File "/usr/bin/pijuice_cli.py", line 269, in main
    status_args = self.get_status()
  File "/usr/bin/pijuice_cli.py", line 215, in get_status
    status = pijuice.status.GetStatus().get('data', {})
AttributeError: 'NoneType' object has no attribute 'status'

Checking
root@DietPi:~# ls -l /dev/i2c*
crw-rw---- 1 root i2c 89, 1 Dec 30 22:10 /dev/i2c-1

Checking
root@DietPi:~# ls -l /usr/bin/pijuice_cli*
lrwxrwxrwx 1 root    root        22 Dec 30 10:41 /usr/bin/pijuice_cli -> /usr/bin/pijuice_cli32
-rw-r--r-- 1 pijuice pijuice 101552 May 16  2021 /usr/bin/pijuice_cli.py
-rwsr-xr-x 1 pijuice pijuice   9096 Dec 17  2018 /usr/bin/pijuice_cli32
-rwsr-xr-x 1 pijuice pijuice  14872 Jan 30  2022 /usr/bin/pijuice_cli64

Please advise how to resolve.

tvoverbeek, 2022-12-31:
Try to run this simple python script in a terminal window:
```
from pijuice import PiJuice

pj=PiJuice(1, 0x14)
ret = pj.status.GetStatus()
print(ret)
```
See if you get any error message that might help

cserl, 2022-12-31:
Thank you for the fast reply. I am powering the setup via official rpi power supply plugged in to the pi zero.

root@DietPi:/usr/bin/python# python3 pijuicezerotest.py

{'data': {'isFault': False, 'isButton': False, 'battery': 'CHARGING_FROM_5V_IO', 'powerInput': 'NOT_PRESENT', 'powerInput5vIo': 'PRESENT'}, 'error': 'NO_ERROR'}

cserl, 2022-12-31:
And after having executed the script from the /usr/bin/python directory, I have tried one again to run pijuice_cli from the same directory. Strangely enough, now it seems to work.
Why does it work now and why did it not work before? In other words what can be the potential risk

tvoverbeek, 2023-01-01:
I do not know the internals of DietPi. Does it by chance have several versions of python (e.g. python2.x, python3.9, python3.10) installed? Maybe it did not find the i2c module (smbus) and hence could not communicate with the Pijuice, How do you install Raspberry Pi Os packages in DietPi?

cserl, 2023-01-01:
As far as I know DietPi is not an own Operating System or Distribution. DietPi is a bundle of scripts on top of a standard Debian installation. Depending on my SBC, DietPi is using different base images. Mainly used images are Armbian, Meveric or as in my case: for PRi boards Raspberry OS.

cserl, 2023-01-01:
With the help of dietpi-software config tool I have installed following software packages:
-Domoticz
-WiringPi  GPIO library
-OpenSSH Server
-Python3 runtime system, package installer
-Python3 Rpi.GPIO

on cli I have installed:
apt-get install pijuice-base

After unsuccessful test result I have installed 
Pijuice from the dietpi-software config tool.

cserl, 2023-01-02:
Firmware version is 5. 
Is there a release history?
Does it make sense to update?

tvoverbeek, 2023-01-02:
For firmware and software release history see https://github.com/PiSupply/PiJuice/blob/master/Software/Source/debian-base/changelog
This file is also part of the installed package. Look in /usr/share/doc/pijuice-base (at least in RasPi OS).
I do not have any data on minimum battery level. With a powerbank you can have a minimum load issue, since they are designed for phone charging.