PiSupply/PiJuice #938: Removing via deb package does not remove all files

nettnikl, 2022-12-31:
Hi!

I'm trying to install this software on another OS, so i was looking a bit into your installer scripts.

The postinst and postrm script do not fit together.
You are creating a symbolic link to another path you are removing it from then, as far as i can see.

### Workaround
Remove it manually

### Quick fix
Adding ` /usr/lib/python3.9/site-packages/pijuice.py` to postrm.

### Correct fix
Similar, but find the installed python version, eg. 3.9 and not assume a specific version.

### Code
postinst
```
# Create symbolic link in /usr/lib/python3/dist-packages so the pijuice module is found by python 3.x (x > 5)
ln -sf /usr/lib/python3.9/dist-packages/pijuice.py /usr/lib/python3/dist-packages  # Build on Bullseye, Debian 11

```
postrm
```
# Remove installed symbolic links
if [ "$1" = "remove" ]; then
	if [ -L /usr/lib/python3/dist-packages/pijuice.py ]; then
		rm /usr/lib/python3/dist-packages/pijuice.py
	fi
```

nettnikl, 2022-12-31:
Okay, i see i have misunderstood: This should be automatically removed, right? It might not be, because i'm running a different python version.

tvoverbeek, 2022-12-31:
Yes,  postrm does remove the symbolic link (even if it points to a different python version)
Closing the issue

closed 2022-12-31 by tvoverbeek