PiSupply/PiJuice #941: Charge 12000mAh 3,7V battery with 12V solar panel

mpourossell, 2023-01-18:
Hi all,

I'm powering the PiJuice Zero using the 12000mAh battery from PiJuice from the J2 connector. To feed this battery I'm using a 20W 12V solar panel connected to a buck converter (12V to 5.1V) to be able to connect it to the J4 pins, which accept from 4,2V to 10V input (so PiJuice does like a solar controller). With this setup, the RPi works for 4 days but battery level keeps dropping untill battery is empty. According to my calculations, I would expect that a 20W panel was way enough powerful to power the system and keep battery levels stable though...

As PiJuice official panels are 5V and are connected to MicroUSB instead of J4 connectors, I'm thinking that the problem might be somewhere in the buck converting from the 12V solar panel to 5.1V, that might cut the Amperage somehow and doesn't provide sufficient current to both power the system and charge the battery (althought the blue light is meaning that battery is charging). Has anyone tried this out already? I didn't find any discussion about how to use 12V solar panels in PiJuice.

Thanks in advance for your help!

xaja34, 2023-05-22:
Were you able to resolve the problem? 
I was hoping to use the same setup.

mpourossell, 2023-05-29:
Using a **thicker wire after the buck converter** solved the issue! I was using current **jumping wires 30awg** from BC to the J4 connectors from PiJuice and the plastic end of the jumper wires was kind of melted after a few days. I soldered **20awg wires instead** to the female termination of the jumper wire and that solved the problem. 

It seems that the current entering to the J4 pins was overheating the wires and made them to lose the connection to the J4 pins after a few days, because of the melted wire. I've been using this system now for almost 1 month in the field and it's prefectly working!

I hope this was useful!