PiSupply/PiJuice #945: GPIO power input NOT_PRESENT when rpi is powered by USB-C

ph818, 2023-02-02:
This is on Raspios64
Firmware updated (1.6), 2021_09_10
$ uname -a
Linux raspberrypi 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux

What I expect:
Status for "GPIO power input" will read PRESENT when the RPI is on and powered by a USB-C power supply (5.0V, 3.0A)

What happens instead:
It reads NOT_PRESENT. If I attach the battery, the rpi begins draining the battery, and if I remove the battery, the RPI loses power and then starts back up.

What I think is happening: the PiJuice is not able to charge the battery from the GPIO pins

Any ideas for how to troubleshoot this?
Thanks!

Patrick

ph818, 2023-02-02:
[image]

ph818, 2023-02-02:
[image]
With battery plugged in, the raspberry pi draws current from the battery, instead of sending current to charge the battery. The LED is green. This time, removing the battery did not cause the RPI to lose power.

ph818, 2023-02-05:
Relevant portions of dmesg:

`i2c-bcm2835 fe804000.i2c: i2c transfer timed out`

if I try changing config.txt to use `dtoverlay=i2c-bcm2708` I don't get that message, but GPIO power input is still not detected.

I have also noticed the following:

* `hwclock -r` always fails with ```hwclock: Cannot access the Hardware Clock via any known method.
hwclock: Use the --verbose option to see the details of our search for an access method.```
* `/proc/device-tree/hat` never exists
* `i2cdetect 1` returns 
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

Let me know if there is anything else I should try.

tvoverbeek, 2023-02-05:
Obviously I2C communication fails.
With the eeprom at 52 i2c comminication with the RTC (0x68) seems to work.
Can you force the VPU core clock to a constant speed (so the i2c clock is also constant)?
See https://www.raspberrypi.com/documentation/computers/configuration.html#mini-uart-and-cpu-core-frequency
Also the i2c clock is derived from the VPU core clock.

ph818, 2023-02-05:
> Obviously I2C communication fails. With the eeprom at 52 i2c comminication with the RTC (0x68) seems to work. Can you force the VPU core clock to a constant speed (so the i2c clock is also [...]

Thank you Ton,

I am trying the configuration
* EEPROM at 00x52
* Power attached to both Raspberry Pi 4 (usb-c), and PiJuice (microUSB)
* BP7X battery installed
* No other peripherals
* Connecting with Wifi
* dtoverlay=i2c-bcm2708
* dtoverlay=i2c-rtc,ds1307
* enable_uart=1
* core_freq=250

I am still getting GPIO Power Input: 5.077 V, 3.577 A, NOT_PRESENT (I expect PRESENT). I can unplug the microUSB and it will switch from PRESENT and CHARGING_FROM_IN to NOT_PRESENT (battery) and NORMAL (battery)

Is it expected that I never see `/proc/device-tree/hat` or `i2cdetect 0` even with EEPROM at 00x50?

entire dmesg here: https://paste.debian.net/1269758/

```
ph818@patpi4raspios64:~$ sudo dmesg | grep i2c
[    4.989805] i2c_dev: i2c /dev entries driver
[    6.647181] bcm2708_i2c fe804000.i2c: BSC1 Controller at 0xfe804000 (irq 22) (baudrate 100000)
[    6.730578] brcmstb-i2c fef04500.i2c:  @97500hz registered in polling mode
[    6.733986] brcmstb-i2c fef09500.i2c:  @97500hz registered in polling mode
ph818@patpi4raspios64:~$ sudo dmesg | grep rtc
[    7.269902] vc4-drm gpu: bound fe206000.pixelvalve (ops vc4_crtc_ops [vc4])
[    7.270270] vc4-drm gpu: bound fe207000.pixelvalve (ops vc4_crtc_ops [vc4])
[    7.270602] vc4-drm gpu: bound fe20a000.pixelvalve (ops vc4_crtc_ops [vc4])
[    7.270899] vc4-drm gpu: bound fe216000.pixelvalve (ops vc4_crtc_ops [vc4])
[    7.271241] vc4-drm gpu: bound fec12000.pixelvalve (ops vc4_crtc_ops [vc4])
[    7.314907] vc4-drm gpu: [drm] Cannot find any crtc or sizes
[    8.425691] rtc-ds1307 1-0068: registered as rtc0
[    8.428351] rtc-ds1307 1-0068: setting system clock to 2023-02-05T17:20:02 UTC (1675617602)
[   17.376353] vc4-drm gpu: [drm] Cannot find any crtc or sizes
```

Thank you for any ideas

tvoverbeek, 2023-02-05:
A few remarks:
- With the EEPROM address at 0x52 the OS will not read the EEPROM (expects it at 0x50) and hence no /proc/device-tree/hat.
- With both Pi and Pijuice powered I see an (estimated) current of 3.577A. Positive current is flowing from PiJuice to the Pi via the GPIO 5V pins.
  It seems your Pi has some power hungry peripherals.
- With only the Pi powered it seems the Pi power supply has not sufficient capacity to also charge the PiJuice battery hence tje NOT_PRESENT and NORMAL.
- With that load the BP7X will not have sufficient capacity to run the Pi. A Pi4 lightly loaded can barely be handled by the BP7X. You need a larger capacity battery (if my assumption about the power needed are correct).
- I currently run my PiJuice on 32-bit. Will have to install the 64-bit OS, but do not know yet when I get around doing that and test my PiJuice with 64-bit.

ph818, 2023-02-05:
Unfortunately I do not see /proc/device-tree/hat even with EEPROM set to 00x50. I can post dmesg of that.

That is a good tip about the Pi drawing a lot of power. There are no peripherals connected: I can take a picture when I get home. But perhaps that is a clue to whatever is breaking. Somehow the 5V 2.5A micro USB cable can power the pi and charge the battery at the same time, but the 5V 3.0 A usb-c power adapter shows up as GPIO Power Input: NOT_PRESENT, with or without a battery. (As in the first screenshot).

I can try raspios32 as well. Will let you know how that goes. My eventual goal is to use PiJuice as a UPS and output Kodi on HDMI so I probably don't need 64 bit for that.

ph818, 2023-02-06:
No hungry peripherals. Here is a picture

[image]

Booting with the configuration:
* EEPROM at 0x50
* Power attached to both Raspberry Pi 4 (usb-c), and PiJuice (microUSB)
* BP7X battery installed
* No other peripherals
* Connecting with Wifi
* dtoverlay=i2c-bcm2708
* NO (#dtoverlay=i2c-rtc,ds1307)
* enable_uart=1
* core_freq=250

Findings:
```
$ sudo dmesg | grep i2c
[    4.834515] i2c_dev: i2c /dev entries driver
[    6.905511] bcm2708_i2c fe804000.i2c: BSC1 Controller at 0xfe804000 (irq 22) (baudrate 100000)
[    6.919185] brcmstb-i2c fef04500.i2c:  @97500hz registered in polling mode
[    6.924623] brcmstb-i2c fef09500.i2c:  @97500hz registered in polling mode```

```
$ ls /proc/device-tree/hat
ls: cannot access '/proc/device-tree/hat': No such file or directory```

```
$ i2cdetect 0
Error: Could not open file `/dev/i2c-0' or `/dev/i2c/0': No such file or directory
$ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

I started fresh with Raspios32, with the procedure described #334 , remove pijuice hardware, fresh install of raspios32, adjust config.txt to include `dtoverlay=i2c-bcm2708` , `enable_uart=1`, and `dtparam=i2c_arm=on`, install pijuice-base, restart, and then attach pijuice hat.

Findings:
`GPIO power input: 5.077V, 3.046A, NOT_PRESENT` *notice it is using .5 A less now*.
`ls: cannot access '/proc/device-tree/hat': No such file or directory`
```
$ i2cdetect -y 0
Error: Could not open file `/dev/i2c-0' or `/dev/i2c/0': No such file or directory
 $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

Do I need to do something to the EEPROM so the HAT is detected correctly? I cannot follow the [EEPROM](https://github.com/PiSupply/PiJuice/tree/master/Firmware/EEPROM) procedure since I have no /proc/device-tree/hat/uuid

Thanks,
Patrick

tvoverbeek, 2023-02-06:
Which OS are you running?
Here is the output of `uname -a` on my 32-bit Pi4 with PiJuice (no peripherals except the PiJuice HAT):
```
tvo@rpi32-deb11:~ $ uname -a
Linux rpi32-deb11 5.15.84-v7l+ #1613 SMP Thu Jan 5 12:01:26 GMT 2023 armv7l GNU/Linux
tvo@rpi32-deb11:~ $ 
```
You do not need the `dtoverlay=i2c-bcm2708` in /boot/config.txt
This was only needed on Buster (debian 10) when the Pi4 was just available.
The only thing which is needed for i2c in /boot/config.txt is `dtparam=i2c_arm=on`.
This line is added/enabled by the pijuice-base installation, so it should already be there.
```
tvo@rpi32-deb11:~ $ dmesg | grep i2c
[    3.812851] i2c_dev: i2c /dev entries driver
[    5.910580] brcmstb-i2c fef04500.i2c:  @97500hz registered in polling mode
[    5.942646] brcmstb-i2c fef09500.i2c:  @97500hz registered in polling mode
tvo@rpi32-deb11:~ $ dmesg|grep rtc|grep -v crtc
[    8.897236] rtc-ds1307 1-0068: Unsupported ohm value 0 in dt
[    8.917297] rtc-ds1307 1-0068: 'wakeup-source' is set, request for an IRQ is disabled!
[    8.942860] rtc-ds1307 1-0068: registered as rtc0
[    8.945721] rtc-ds1307 1-0068: setting system clock to 2023-02-06T12:42:42 UTC (1675687362)
tvo@rpi32-deb11:~ $
```
i2cdetect output:
```
tvo@rpi32-deb11:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
tvo@rpi32-deb11:~ $
```
Note the 'UU', The rtc driver was loaded due to the device-tree fragment in the HAT eeprom.

```
tvo@rpi32-deb11:~ $ ls -l /proc/device-tree/hat/
total 0
-r--r--r-- 1 root root  4 Feb  6 13:54 name
-r--r--r-- 1 root root 12 Feb  6 13:54 product
-r--r--r-- 1 root root  7 Feb  6 13:54 product_id
-r--r--r-- 1 root root  7 Feb  6 13:54 product_ver
-r--r--r-- 1 root root 37 Feb  6 13:54 uuid
-r--r--r-- 1 root root 10 Feb  6 13:54 vendor
tvo@rpi32-deb11:~ $ 
```
Finally 2 screenshots, first with only Pi powered, second with only PiJuice powered.
[image]

[image]

Suggest to start with a clean install of Raspberry Pi OS 32-bit, followed by the usual `sudo apt update` and 'sudo apt upgrade` and reboot.
No changes to /boot/config.txt.
Then `sudo apt install pijuice-base`
Check if `dtparam=i2c_arm=on` is enabled in /boot/config.txt.
The the usual i2cdetect output, dmesg, etc.

If then it does not work, there must be something else going on.

ph818, 2023-02-06:
Thanks, I will try. 
Point of clarification: Doing `rpi-update` is unnecessary, correct? I have not done it due to its warning message.

* First, resetting PiJuice back to default configuration
* Remove PiHat
* Create new RaspiOS32 sd card
* sudo apt update; sudo apt upgrade; sudo reboot

```
 $ uname -a
Linux patpi4raspios32-1 5.15.84-v7l+ #1613 SMP Thu Jan 5 12:01:26 GMT 2023 armv7l GNU/Linux
``` 
(matches yours)

* sudo apt install pijuice-base

``` 
$ sudo cat /boot/config.txt | grep i2c
dtparam=i2c_arm=on
``` 
confirmed present

* (shutdown, remove power)
* Re-attach PiJuice HAT with battery
* Power on (supplied by laptop USB-C power supply, capable of 5.0V 3.0A)

```
 $ dmesg | grep i2c
[    3.628034] i2c_dev: i2c /dev entries driver
[    6.089314] brcmstb-i2c fef04500.i2c:  @97500hz registered in polling mode
[    6.090218] brcmstb-i2c fef09500.i2c:  @97500hz registered in polling mode
``` 
same as you

`dmesg|grep rtc|grep -v crtc` **No Output** (I can see the crtc stuff if I remove the last part)

```
 $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
``` 
consistent with it not seeing any EEPROM at 0x50 (?)

``` 
$ ls -l /proc/device-tree/hat/
ls: cannot access '/proc/device-tree/hat/': No such file or directory
```

Only Pi powered (but battery attached)
[image]
pi appears to be drawing power from the battery instead of USB-C.

Only battery powered
[image]
no change

Both Pi and MicroUSB-Powered (battery attached)
* CLI initially shows CHARGING_FROM_IN / NOT_PRESENT / PRESENT
* pijuice_cli crashes after 1 second or so
 
```
$ pijuice_cli
Traceback (most recent call last):
  File "/usr/bin/pijuice_cli.py", line 2087, in <module>
    loop.run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 287, in run
    self._run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 385, in _run
    self.event_loop.run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 790, in run
    self._loop()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 823, in _loop
    alarm_callback()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 173, in cb
    callback(self, user_data)
  File "/usr/bin/pijuice_cli.py", line 259, in update_status
    status_args = self.get_status()
  File "/usr/bin/pijuice_cli.py", line 224, in get_status
    general_info += ", %3d°C, %s" % (temp, status['battery'])
KeyError: 'battery'
```

Restart pijuice_cli, only MicroUSB power (battery attached)
[image]

Re-attach USB-C power to the Pi (battery attached)
[image]
(notice Pi still seems to be draining power from the battery, not using USB-C)

At this point I expect if I remove the battery, the Pi will crash, since it is not drawing power from USB-C
Battery Removed: Pi loses power and immediately regains it and restarts.
[image]
(What values to you get with Pi powered, no battery?)

Findings:
* Pi still does not detect the EEPROM at 0x50, does not load rtc
* GPIO power still NOT_PRESENT in `pijuice_cli` with pi powered by USB-C. Draws power from battery instead
* Pi loses power when PiJuice loses power, even with USB-C is connected. But immediately regains power and restarts.
 
Is this a hardware problem with the PiJuice HAT? Would `rpi-update` change anything (and is worth the risk?)

tvoverbeek, 2023-02-07:
Patrick,

Last resort:
Check if there is content in the HAT eeprom using eepflash to dump the eeprom and eepdump to make the dump human readable.
Also have you tried to reflash the firmware using pijuiceboot?

If thst does not work you probably have a defect PiJuice.

ph818, 2023-02-07:
Hi Ton,

Yes, I first upgraded the firmware from within pijuice_cli, and later redid it with `cd /usr/share/pijuice/data/firmware/; pijuiceboot 14 PiJuice-V1.6_2021_09_10.elf.binary`

I am not sure how to read the HAT content with eepflash. From https://forums.raspberrypi.com/viewtopic.php?t=108134 and https://github.com/raspberrypi/hats/tree/master/eepromutils Can you see if this procedure is missing something?

```
$ sudo dtoverlay i2c-gpio i2c_gpio_sda=0 i2c_gpio_scl=1 bus=9
$ i2cdetect -y 9
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
$ sudo hexdump /sys/class/i2c-adapter/i2c-0/0-0050/eeprom
hexdump: /sys/class/i2c-adapter/i2c-0/0-0050/eeprom: No such file or directory
hexdump: all input file arguments failed
```

I tried changing to 0x52 with pijuice_cli, rebooted, and tried again. Still nothing on i2cdetect or hexdump. I also figured out eepflash for this one:

```
$ sudo ./eepflash.sh -r -f=dump.eep -d=0 -a=52 -t=24c32
This will attempt to talk to an eeprom at i2c address 0x52. Make sure there is an eeprom at this address.
This script comes with ABSOLUTELY no warranty. Continue only if you know what you are doing.
Do you wish to continue? (yes/no): yes
./eepflash.sh: 159: cannot create /sys/class/i2c-adapter/i2c-0/new_device: Directory nonexistent
Reading...
dd: failed to open '/sys/class/i2c-adapter/i2c-0/0-0052/eeprom': No such file or directory
Closing EEPROM Device.
./eepflash.sh: 187: cannot create /sys/class/i2c-adapter/i2c-0/delete_device: Directory nonexistent
Error doing I/O operation.
```

Changed eeprom back to 0x50 and tried again. Nothing on i2cdetect -y 9

```
$ sudo ./eepflash.sh -r -f=dump.eep -d=0 -a=50 -t=24c32
This will attempt to talk to an eeprom at i2c address 0x50. Make sure there is an eeprom at this address.
This script comes with ABSOLUTELY no warranty. Continue only if you know what you are doing.
Do you wish to continue? (yes/no): yes
./eepflash.sh: 159: cannot create /sys/class/i2c-adapter/i2c-0/new_device: Directory nonexistent
Reading...
dd: failed to open '/sys/class/i2c-adapter/i2c-0/0-0050/eeprom': No such file or directory
Closing EEPROM Device.
./eepflash.sh: 187: cannot create /sys/class/i2c-adapter/i2c-0/delete_device: Directory nonexistent
Error doing I/O operation.
```

for good measure,
```
$ sudo dmesg | grep i2c
[    3.701070] i2c_dev: i2c /dev entries driver
[    5.612049] brcmstb-i2c fef04500.i2c:  @97500hz registered in polling mode
[    5.614338] brcmstb-i2c fef09500.i2c:  @97500hz registered in polling mode
[   54.283813] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /aliases/i2c_gpio
[   54.283851] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /__symbols__/i2c_gpio
[   54.304679] i2c-gpio 900000000.i2c: using lines 0 (SDA) and 1 (SCL)
```

tvoverbeek, 2023-02-07:
Look at the source of eepflash.
The OS has already claimed i2c bus 0 (GPU + camera).
That is why -d=0 does not work.
When you do not specify a bus with -d, eepflash creates a bit-banged i2c controller on bus 9.
Also you do not have to specify the eeprom address. It defaults to 50.
Try with `sudo ./eepflash.sh -r -f=dump.eep -t=24c32`

ph818, 2023-02-08:
Thanks, I tried with EEPROM at 0x50, and at 0x52, and it failed to see it each time.

```
$ sudo ./eepflash.sh -r -f=dump.eep -t=24c32
This will attempt to talk to an eeprom at i2c address 0x50. Make sure there is an eeprom at this address.
This script comes with ABSOLUTELY no warranty. Continue only if you know what you are doing.
Do you wish to continue? (yes/no): yes
Reading...
dd: failed to open '/sys/class/i2c-adapter/i2c-9/9-0050/eeprom': No such file or directory
Closing EEPROM Device.
Error doing I/O operation.
s$ sudo ./eepflash.sh -r -f=dump.eep -a=52 -t=24c32
This will attempt to talk to an eeprom at i2c address 0x52. Make sure there is an eeprom at this address.
This script comes with ABSOLUTELY no warranty. Continue only if you know what you are doing.
Do you wish to continue? (yes/no): yes
Reading...
dd: failed to open '/sys/class/i2c-adapter/i2c-9/9-0052/eeprom': No such file or directory
Closing EEPROM Device.
Error doing I/O operation.
```

At this point I sadly conclude that the pijuice has some hardware defect.

Thank you for your help troubleshooting this. I know significantly more about electronics and the pijuice now at least.