PiSupply/PiJuice #951: Pijuice is "clicking"

frohr, 2023-02-14:
Hi, I have RB 4b and Pijuice with 12000mAh battery. I have connected Teensy 4.0 and adxl1005 via USB to RB to collect acceleration data. It was ok, but now pijuice started "clicking" (I can hear it) and it interference somehow with USB and I see big peaks every 100ms in my acceleration data. This issue is with battery and without battery and external power to pijuice. If I power RB directly, it is ok.
Any experience with this issue?
Thanks

frohr, 2023-02-16:
Here is picture of peaks from Pijuice:
https://pasteboard.co/bjq67IkrMHsy.jpg

hcfman, 2023-07-09:
Ah yes, I have the same problem. For me it's audio recording application so clicking is not an option.