PiSupply/PiJuice #952: Buttons - HARD_FUNC doesn't work when using POE HAT

proggprod, 2023-02-18:
My PiJuice HAT doesn't respond to button presses for HARD_FUNC_POWER_ON, HARD_FUNC_POWER_OFF, or HARD_FUNC_RESET when  power is being supplied by POE.

My POE hat is between the Pi4 and the PiJuice.

If the cable is in and supplying power I cannot use the HARD_FUNCs.
To turn off, I have to unplug the ethernet cable first, then press the buttons. 
And then pressing buttons _without_ the ethernet plugged in, OR plugging in the ethernet cable restores power.

If the Pi happens to be off with the ethernet still connected, I cannot turn on with the buttons. I have to re-plug the ethernet, or unplug the ethernet and press a button, then reconnect ethernet.

Everything else seems to work fine

tvoverbeek, 2023-02-18:
You have 2 HATs supplying power in parallel via the GPIO pins to the RPi.
The PiJuice HARD_FUNC_* functions will not work since power is still supplied by the POE HAT.
If you consider the 2 parallel power sources then it explains exactly what you are seeing.

closed 2023-02-18 by tvoverbeek

proggprod, 2023-02-18:
OK. As I'm experimenting, I agree with you, it doesn't seem to be a 'bug'. But it would still be nice if the buttons worked as expected without a magic dance.

proggprod, 2023-02-19:
The consequence of this is that although the PiJuice can help with shutting down when the battery gets low, I don't think it will be able to reboot when charge comes back. And it wont be able to reset in the event of a system crash because both of these functions require HARD_FUNC, and in these scenarios there could be POE available at the time

It would be nice to find a work around or solution to make it possible to use POE and still retain the benefits/features of the PiJuice

tvoverbeek, 2023-02-19:
Stumbled over this: https://thepihut.com/products/usb-c-gigabit-poe-splitter-5v-2-4a
This way the Pi4 is powered via its 'normal'USB-C connector and there is no parallel feeding via the GPIO.
Should work fine with the PiJuice

proggprod, 2023-02-20:
I wonder if the ethernet on this passes though the power anyway, resulting in power going to POE HAT and USB C. It doesn't say on the product page. Or are you suggesting that I remove the POE HAT to be replaced by this splitter?

tvoverbeek, 2023-02-20:
Yes, replace the POE Hat with the splitter.