PiSupply/PiJuice #953: DATA_CORRUPTED on RPi4 and software v1.6

kingswindsor, 2023-02-20:
Hi, I have a Pi4B with Bullseye installed and have been using it to run an MLX90640 IR camera showing the output on a small HDMI screen connected to it.  Phase 2 of the project has been to get it working with PiJuice but I have encountered the DATA_CORRUPTED error.  Reading back, I have tried modifying config.txt as suggested but that didn't help and that is probably unnecessary as I am using software at v1.6.
Grateful for any advice.  I have already modified config.txt for the MLX90640 camera and it is connected to the GPIO pins so maybe some conflict?  But how can I resolve? thanks

tvoverbeek, 2023-02-20:
What frequency are you using for the i2c clock? If more than 400kHz (i.e. 1 mHz as mentioned in the Pimoroni docs) then you twil ave problems with the PiJuice.
There is no address conflict. The MLX90640 uses 0x33 (as far as I know), the PiJuice 0x14 and 0x68

kingswindsor, 2023-02-21:
Thank you, I have checked the addresses and these addresses are being used by the MLX90640 and the PiJuice 0x33, 0x14 and 0x68

 I'm not sure how to definitively check the i2c clock frequency (apologies) but the /boot/config.txt file includes

[all]
dtparam=i2c_arm=on
dtparam=i2c1_baudrate=400000
gpu_mem=128

Should I throttle back on that baud rate?

tvoverbeek, 2023-02-21:
You can try to go back to 100000.