PiSupply/PiJuice #955: SW1 Not working after Factory reset

dmkrueger67, 2023-03-02:
Hello, My Pi Juice seemly went dead after leaving the battery disconnected for days. I needed to factory reset to get it to do anything. After Resetting, the SW1 switch doesn't turn the Pi on, I have even hooked an external switch to the SW1 terminals and it does nothing as well. The unit will still wake at the time I specified, I just can't turn it on with a switch anymore. I have also noticed the light is solid green, even when unplugged.

Any help would be appreciated. I did see another open issue, but it had no resolution.

Thanks
Dean Krueger

tvoverbeek, 2023-03-02:
How did you do factory reset?
Anyway, suggest to reload the firmware with pijuiceboot.

dmkrueger67, 2023-03-02:
I held the two left most buttons until I saw the light come on. Currently the only way I can turn on the pi is by applying power directly to it. I can get into the PiJucies command line configure app once I plug in the Pi.

I can try the firmware, what is the process?

Thanks for your response.

tvoverbeek, 2023-03-02:
The two left buttons are SW1 and SW3
The Switches are in the order SW1-SW3-SW2.
So I do not know wht the effect is of pressing SW1 and SW3 simultaneoulsy.
From the documentation:
----
Dual long press of SW1 and SW2 for 20 seconds will reset PiJuice HAT configuration to default. This applies to the MCU configuration only.
----
THis only resets the parameters in the MCU, not the parameters for the system task.

For firmware reload see https://github.com/PiSupply/PiJuice/tree/master/Firmware#pijuice-firmware
Use the latest firmware file (V1.6)
You might have to use the procedure outline  at the bottom of that page: https://github.com/PiSupply/PiJuice/tree/master/Firmware#pijuice-bricked-after-unsucsessful-firmware-update

dmkrueger67, 2023-03-03:
When I look at the unit the left most switch is SW2 then SW3, SW1 seems to do nothing since the reset.
[
[image]
](url)

I read somewhere in this forum to hold the left most buttons, that would factory reset. Prior to this problem, I had the LEDs disabled, when holding those buttons, they started blinking again, that's why I assumed it was reset.

I will attempt to reload the firmware this weekend, at this point its waking and doing the job I need of it, I just can't turn the PI on manually.

tvoverbeek, 2023-03-03:
If you look at the unit (mounted on the RPi) from the short side (the side with the RPi SD-csrd and DSI display connector) then indeed SW2 is the left most one.
Remains, the efect of pressing SW2 and SW3 simultaneously is not defined.
Let us know how it goes with reloading the firmware.

dmkrueger67, 2023-03-05:
I found the problem and everything is good! When soldering the external switch wires, I had a piece of solder bridge the SW1 contacts. As soon as that was removed, everything worked as expected. Refreshing the firmware worked as well.

Thanks for everything, this is a well built product!

tvoverbeek, 2023-03-05:
Glad you found it.
Closing the issue

closed 2023-03-05 by tvoverbeek