PiSupply/PiJuice #961: Swollen battery

rascalos, 2023-03-24:
Hi,
We are having BP7X battery issues on several PiJuice. They swell which sometimes tends to cause problems with the contacts of the PiJuice PCB.
Do you know if there are parameters to monitor to know the state of the battery which would allow you to know when to change the battery in advance?
Moreover, given the feedback you have on this type of problem, is there a battery that has fewer problems? if yes which one ?
Regards,

tvoverbeek, 2023-03-24:
A swollen battery is always bad. It means there is internal damage in the battery and it should be replaced asap.
Note the BP7X was originally intended for the Motorola flip-phones (early 2000s) so depending on where you bought it, the bp7x could have had a long shelf life.
Just FYI, I had to replace one BP7X because it could not hold charge (any load and the voltage dropped)

rascalos, 2023-03-27:
Hi,
Thanks for your feedback. 
Are there parameters that should be monitored that could announce battery problems?
If so, which ones and how?
Regards,