PiSupply/PiJuice #966: Curiosity - is USB C support planned?

rdragonrydr, 2023-04-22:
I've been watching your catalogue for a while, I've been somewhat curious if this would really work with the Raspberry Pi 4 - my current unit is still deployed to a Pi 3 and I haven't wanted to possibly break it if there's too much current draw.

Other than the current, though, I noticed that you're still selling this with micro USB and have not changed the connector. I'd prefer to get one with a matching USB C connector if you're going to release an updated board revision any time soon.

tvoverbeek, 2023-04-22:
I cannot answer that since I am not employed by PiSupply/Nebra. I am only an interested volunteer.
I have not seen an entry by PiSupply here for a long time.
My suspicion is they are concentrating on their Helium miners (www.nebra.com).
Also they have been affected by the component shortage and high prices for the STM32 MCU.
There were plans for a higher capacity PiJuice version, but nothing has shown up yet.
Try to contact PiJuice/Nebra  directly.

OhSoGood, 2023-06-07:
I will add my two cents to what @tvoverbeek wrote: without him (not a PiSupply staff), there would nearly be no support on Pijuice. The interest of Nebra / PiSupply in the Pijuice seems to have declined