PiSupply/PiJuice #971: COMMUNICATION_ERROR

PUW149, 2023-05-07:
I have an Rpi3B+ with a PiJuice HAT, nothing else on the GPIO. Everything was working fine and I was able to use the PiJuice software (GUI and CLI), but then started to get "COMMUNICATION_ERROR" in the PiJuice software. I have tried a couple of batteries in the PiJuice in case that was the problem.
I2C is enabled and I get:
pi@raspberrypi:/boot $ lsmod | grep i2c
regmap_i2c             16384  1 rtc_ds1307
i2c_bcm2835            16384  1
i2c_dev                20480  2

and:
pi@raspberrypi:/boot $ ls -l /dev/i2c-1 
crw-rw---- 1 root i2c 89, 1 May  6 09:40 /dev/i2c-1

predictably given the GUI error, 
i2cdetect -y 1

gives nothing.

I am unable to flash the firmware (predictably as there is no I2C connection) 
""error receiving data -1" when trying to update firmware"

If I check voltages with a multimeter 

GPIO 2 gives 3.3V
GPIO 3 gives 0.2V

both with PiJuice attached, and with it detached.

checking the GPIO pins gives:

pi@raspberrypi:/boot $ raspi-gpio get
BANK0 (GPIO 0 to 27):
GPIO 0: level=1 fsel=0 func=INPUT
GPIO 1: level=1 fsel=0 func=INPUT
GPIO 2: level=1 fsel=4 alt=0 func=SDA1
GPIO 3: level=0 fsel=4 alt=0 func=SCL1
....

There is nothing unusual in config.txt 

Questions:
(i) Am I right in assuming that this means there is a problem with the RPi GPIO Bus?
(ii) Can I set other Pins to be used as I2C rather than GPIO 2 and 3, and then get PiJuice to communicate using those pins? I am assuming that this is impossible as I can't communicate with the PiJuice to tell it to use alternate pins
(iii) is it worth me trying a new SD card with a clean installation? or is this definitely the RPi hardware

Thanks,
Simon

tvoverbeek, 2023-05-07:
The I2C clock frequency varies with the Core frequency. Under load the core freq goes up and so does the I2C clock freq.
I have seen this causing communication error.
You can fix the core freq in /boot/config.txt.
See the discussion in this issue: https://github.com/raspberrypi/linux/issues/4123
Suggest to first try with 'enable_uart=1' in /boot/config.txt

PUW149, 2023-05-07:
Thanks Ton.
I have tried (separately):
enable_uart=1
dtparam=i2c_arm_baudrate=10000
and
dtoverlay=i2c-bcm2708

on /boot/config.txt, but no success yet
Is there a recommended baudrate for i2C on the PiJuice?

tvoverbeek, 2023-05-07:
Note: standard baudrate for I2C is 100000 (100kHz), not 10 kHz

PUW149, 2023-05-07:
I was trying it  slower to see if that helped. Trying with a fresh SD card and vanilla Rpi Bullseye OS now.

tvoverbeek, 2023-05-07:
First thing to check wrt I2C is if i2cdetect works (of course with something connected to the I2C bus).

PUW149, 2023-05-07:
Nope, still nothing on I2C.  Could be time for a new Rpi. Or could this be the PiJuice?

tvoverbeek, 2023-05-07:
Do you have an other I2C peripheral which you could try?

PUW149, 2023-05-07:
The only other I2C I have is an Element 14 Cirrus Logic Audio Card which I haven't used for years. My Rpi can't see that on I2C either, which supports this being an Rpi problem rather than a PiJuice problem