PiSupply/PiJuice #977: Swollen batteries (24/7 use case)

jonnyns, 2023-05-24:
I have several (in fact 7) Raspberry Pis with PiJuice UPS hats. Their intention is to work 24/7 for various tasks.
They have now been constantly working for more than one year, but - on inspection - I have noticed that several (at least 3) seem to experience severe battery swelling.
As can be seen from the picture the batteries (original, delivered with the hardware) are marked 20200615, so they are not really *that* old.

[image]

I tend to keep the PiJuice firmware up-to-date, now at V1.6.

As the systems are running 24/7 and also connected to mains power, the batteries appears to be charged to 98% most of the time.

I wonder if battery swelling is a general problem with PiJuice, due to either 
- the specific battery manufacturer used for original batteries,
- the fact that batteries are kept top charged for most of the time,
- some setting for the PiJuice hat that should be altered for such a use case.

I have noticed that there are some other reports on battery swelling, but to me this starts to look as a general PiJuice issue. It seems pretty strange to me if this is not experenced by quite some of PiJuice users...
Any thoughts/experiences related to this???

chrisbtmn, 2023-05-28:
My experience is similar.

Mine runs 24x7 on mains power with the PiJuice for backup power and only rare power interruptions.

I'm on my third swollen BP7X battery now. The latest was new in October 2022 and needs replacing now.

I've noted in the past that the temperature seems too high. The PiJuice software reports about 25C without a battery but about 50C with a battery even when it's not charging or providing power.

I don't want to go on replacing the battery about every 6 months.

I have wondered about using a different type of battery but I don't know if that would be any better (and it wouldn't be as neat).

tvoverbeek, 2023-05-28:
Probably the constant trickle charging has something to do with the battery swelling.
You could try to ony charge till the battery is say 90% full and then disable charging.
Re-enable charging when the chargelevel reaches say 20%.
See the functions SetChargingConfig and Get ChargingConfig in pijuice.py lines 1022-1050.

jonnyns, 2023-05-29:
@chrisbtmn Thanks for sharing your experiences.

Although I have see some few complaints about PiJuice battery swelling, I am surprised that I have
not seen **much** more, as I now feel pretty confident that this is a general PiJuice problem, at least for the 24/7 use case. In my view this should be a warning flag for potential PiJuice buyers.

As I am unhappy about the situation, I wonder if I should investgate more into changing to another battery chemistry, LiFePo4, seems to be a very viable solution. And safer as well, according to what I can tell.

I use a 3D printed stacked design, so I can easily print a matching side tray for an 18650 battery. It will not be as neat as using an "internal" BP7X battery, as the battery will be connected to the PiJuice via wires, implying that I cannot just slide out the RPi assembly without disconnecting the wires first. But it should work.
[image]

jonnyns, 2023-05-29:
@tvoverbeek Thanks for your comment.

You might be very right about "trickle charging" being part of the problem.
I take it thought, that the 24/7 use case battery swelling is not experienced by you, at least not yet?

In my view, the charging/discharging conditions shoud be part of the common gui setup, to avoid that everyone should
need to fiddle with the source code? The need for this should be obvious enough, I think?

I have observed a previous "issue thread" on LiFePo4 usage with LiFePo4 chemistry. As you seem to have much more knowledge on this matter that I currently do, is there any special information you could share on using LiFePo4 with PiJuice before I dive further into the matter?

Thanks in advance.