PiSupply/PiJuice #980: COMMUNICATION_ERROR with PiJuice Zero with Pi Zero

ch-lee, 2023-06-01:
I think I have tried almost everything and also had a look at related issues:

* https://github.com/PiSupply/PiJuice/issues/647
* https://github.com/PiSupply/PiJuice/issues/610
* https://github.com/PiSupply/PiJuice/issues/506
* https://github.com/PiSupply/PiJuice/issues/268
* https://github.com/PiSupply/PiJuice/issues/175

**Problem:**
The status of the PiJuice Zero is displaying a `CONNECTION_ERROR` status when using the `pijuice_cli`

* I have connected the Pi Zero to the PiJuice using the pins, and the battery is connected to the PiJuice. The LED flashes and alternates between green and blue - see attached images below
* I have enabled I2C Interface by following the steps from https://www.raspberrypi-spy.co.uk/2014/11/enabling-the-i2c-interface-on-the-raspberry-pi/ - see screenshot below
* Any help or suggestions would be greatly appreciated, I'm very new to the raspberry pi community and how the things work. 

**Hardware:**
* Raspberry Pi Zero
* PiJuice Zero
* 12,000mAh Battery

**Screenshots**
Image of PiJuice Zero with GREEN LED
[image]

Image of PiJuice Zero with BLUE LED
[image]

Image of I2C setting enabled
[image]

**Steps and things I have tried:**
Installed pijuice `sudo apt-get install pijuice-base`

I can install `pijuice-gui` on the pi zero. But when I launch the app, nothing happens. I presume the Pi Zero is not powerful enough?

When I run `pijuice_cli`, in the terminal

Enter "status" - shows `CONNECTION_ERROR`

[image]

I tried running `i2cdetect -y 1` and get the following output

```
chuen@raspberrypi:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --     
```

I have noticed that the `i2cdetect` should output as described here: https://github.com/PiSupply/PiJuice/issues/175#issuecomment-407520752

I also ran `lsmod|grep i2c` and the output below

```
chuen@raspberrypi:~ $ lsmod|grep i2c
i2c_bcm2835            16384  1
i2c_dev                16384  2
```

I also tried updating the firmware following these instructions in https://github.com/PiSupply/PiJuice/blob/master/Firmware/README.md
Below is the output
```
chuen@raspberrypi:~/Documents/GitHub/PiJuice/Firmware $ pijuiceboot 14 PiJuice-V1.6_2021_09_10.elf.binary 
using address 14
Input file PiJuice-V1.6_2021_09_10.elf.binary
Starting bootloader 
Starting bootloader old firmware
Sending get command 
error receiving data -1
```

I have also tried installing a fresh raspberry pi OS on another memory card and have the same `COMMUNICATION_ERROR` status.

I haven't tried the "bootloader" as described in https://github.com/PiSupply/PiJuice/issues/647#issuecomment-782040241 as I wasn't sure where the SW3 button is located on the PiJuice Zero. There is only one small button beside the LED on the PiJuice Zero, I presume this is the one?

Any other suggestions? I think I have exhausted everything? I may have a dead PiJuice Zero?? Can I get a replacement?

bryanlegrand, 2024-02-08:
@ch-lee perhaps some hope for you.

I experienced the exact same problem. Did all the exact same steps as you did as well with no help. Here are a few things that got my over the edge..

1. Put the PiJuice on a rpi4, run i2cdetect, can see the pijuice.
2. Flash an sd card with 64-bit bullseye (had to be 64, 32-bit didn't work for upgrading the firmware), installed pijuice gui, restarted, flashed the firmware from 1.5 to 1.6 from the gui. Everything works fine.
3. Put the pijuice zero back on my zero 2 w, and... nothing. Start to move it around a bit hoping for a miracle and... bam, it works.

quite simply the gpio pin that comes with the pi zero 2 w wasn't making good contact with the board... 6 hours of racking my brain trying to understand how/why and it was just that, a not very good physical connection...

ch-lee, 2024-02-10:
Thanks @bryanlegrand for sharing your experience.

I think I have given up all hope since it was radio silence with this issue. Thought I received a dud.

It's rather frustrating its down to bad connecting pins with the board. 

Glad you got it sorted, I may try what you suggested. Thanks again.