PiSupply/PiJuice #983: compatibility options with other SoC board

pedrosk, 2023-06-08:
Since the Raspberry PI is out of stock for the past very long time I am looking on building my systems on other SoC boards. What else is Power HAT compatible with besides Raspberry PI?
There is a similar issue but does not seem to [Compatibility with other hardware (Orange Pi or Odroid) ](https://github.com/PiSupply/PiJuice/issues/831)

I was wondering if there is any success with other boards. 
Also since there seem to be lack of PIs but other boards seem to be available I am wondering if there is any consideration from PiSupply and maybe by community to port these to other SoC boards,

I look forward to hear back.

Maybe please open and don't close

shawaj, 2023-06-09:
The RockPi 4b is pretty reliable, we use it in production for some projects.

However not sure how well the PiJuice HAT would with with it, and you'd have to port the software yourself which would be non trivial.

Hopefully the raspberry pi shortages alleviate soon anyway

pedrosk, 2023-06-09:
@shawaj  If I understand correctly RockPi 4b is a good board but you did NOT use it with PiJuice .

shawaj, 2023-06-09:
> @shawaj If I understand correctly RockPi 4b is a good board but you did NOT use it with PiJuice .

Yes we use RockPi for our Helium Miners and some other stuff... https://www.nebra.com/products/nebra-hnt-indoor-hotspot-miner-rock-pi-version

But haven't tried it with the PiJuice