PiSupply/PiJuice #987: Is project dead? PiSupply is absent fo ages, no support, no update...

OhSoGood, 2023-06-18:
Hi
I really wonder if the project is dead. The PiSupply team has not been seen for ages, there is no support (except by individuals - thanks to them!), there is no update of code despite bugs. I use 12.000mAh batteries, and the firmware is buggy for these (pitifully, deactivating the NTC is not a solution, when device are outside under the sun).

With 100+ Pijuice purchased here, I'm worrried.
I am going to look for alternatives. If anybody has suggestion, they are welcome.

jonnyns, 2023-06-28:
I also wonder if PiJuice project is dead. From my point of view, PiJuice with LiPo batteries cannot be used for 24/7 type of operations, as it tends to eat batteries (they swell).

I have not tried this, but an interesting alternative appears to be the crowd-funded LiFePO4wered/Pi+ project at https://www.crowdsupply.com/silicognition/lifepo4wered-pi-plus.

This site claims that LiPo batteries "are inherently a poor choice for a UPS because they quickly degrade when constantly kept fully charged".

One "drawback" with this solution is that it has a form-factor incompatible with PiJuice, so the enclosure used must be redesigned.

Anyone tried this?
[image]

shawaj, 2023-06-28:
The project isn't dead and you can still get in contact by emailing sales@pi-supply.com

closed 2023-06-28 by shawaj