PiSupply/PiJuice #989: Known working stackable header for Pijuice Zero?

zeitkunst, 2023-07-11:
I'm trying to figure out the best stackable header for my Pi Zero 2 that _actually_ works with the Pijuice Zero. All of the headers that I've tried so far make poor connections with the Pijuice Zero, meaning that no i2c connections can actually be made between the Zero and the Pijuice.

My desire is to put a couple of stackable headers on my Pi Zero 2, thus raising the Pijuice Zero a few centimeters above the Pi Zero 2. Why?

- Having the Pijuice Zero so close to my Pi Zero 2 causes extreme wifi degradation
- I need to put a headsink on my Pi Zero 2, as the board will be partially buried underground and I will be making a large heatsink to disperse the heat into the ambient air above ground

So far, I've tried a stackable header such as this one: https://www.kiwi-electronics.com/en/40-pin-gpio-stacking-header-2x20-female-1640 . The connection between this header and the Pijuice Zero header is quite poor and no i2c connections can be made.

I've tried many things to make this work, including bending the pins of the stackable header so that they make better connections (which didn't work), and attempting to solder the pins to the Pijuice header (which has proven nigh impossible given that there is no easy way to make contact with the metal connections inside the header).

The header on the full-sized Pijuice works perfectly, and in the end, I may just use the full-size board with my Pi Zero 2 because I don't have these i2c connection problems with it, and I don't have wifi degradation. But, I'd rather not have wasted money on the Pijuice Zero if there is a way to make it work.

Does anyone have suggestions of a stackable header that has worked for them? Thanks in advance!

tvoverbeek, 2023-07-11:
First is this a Pi Zero 2 WH (with headers) or did you solder a 40 pin header yourself?

Did you check if I2C works without the stackable header(s)? You can use jumper wires to connect
the PiJuice Zero to the Pi Zero 2 or connect an other i2C peripheral.
If you soldered it yourself you could have a bad solder connection on the I2C pins. In that case no
stackable header will work reliably.

PS Since you mention KIWI electronics, are you located in the Netherlands?

zeitkunst, 2023-07-11:
I soldered the headers myself, but this same Pi Zero 2 works just fine with my full-size Pijuice, so it's not a soldering problem with the headers on the Pi Zero 2.

Yes, the Pijuice Zero works without stackable headers, but leads to the issues I mentioned in the original post (poor wifi, no ability to install a heatsink).

I'm not sure how to solder jumper wires between my Pi Zero 2 headers and the Pijuice Zero, since the header on the Pijuice Zero doesn't really provide much space to do so?

And yes, I'm based on the Netherlands :)