PiSupply/PiJuice #992: Pi Zero W not turning on

bjohas, 2023-07-19:
We bought 10 PiJuice Zero units, and are struggling with configuration of some aspects. The documentation is comprehensive, though we couldn't quite make sense of it in some places. Generally speaking, we have a problem with the PiJuice waking up a pi zero w.

Here are some areas, where we hope you can provide some more information.

**1. Wakeup alarm**

1a. The PiJuice Zero has a wake-up alarm. We are using a pi zero w, which is powered via gpio from the PiJuice (stacking header). Power is via the usb port on the pijuice. We've set the alarm for hour 9, every day, minute 0. Our expectation was that the device would turn on at that time, but it didn't. Do you have any insights on this? 

(We do have the recommended crontab script running.)

1b. Also, should the wakeup happen irrespective of the battery charge status? If the battery status is below the shutdown value set (say threshold set to 30% and battery charge is 25%), then should the wakeup alarm still work? And if yes, is the low battery shutoff then ignored?

1c. Could you explain what the difference is between "minute" and "minute period"?

**2. The heartbeat monitor.** 

Could you explain how this works? Does the PiJuice just do it all automatically or are there other settings needed? Generally speaking, when the zero becomes unresponsive, the heartbeat monitor doesn't seem to reboot the pi. 

Related to this, we've found that the zero sometimes doesn't reboot. I.e., we do a sudo reboot and the Pico stays off. When we then use the button on the PiJuice, the device turns on. 

**3. Boot when charge threashold is reached.**

The boot when the battery threshold is reached isn't working. We've not been able to test this extensively, so not sure about this entirely.

Overall, this seems to point at the PiJuice not being able to wake up the zero. 

So - what's going wrong? Do we have the connection wrong? How should we connect the PiJuice?

tvoverbeek, 2023-07-19:
Bjoern (or is it Björn?) here a few answers
1a The alarm time is in UTC, not local time. So if you are in e.g. Germany then hour 9 means 11 local (summertime UTC+2).
1b The PiJuice will try the wakeup (provided it is enabled), but as soon as the PiJuice service starts on the Pi it will detect the belo threshold and shut down. 
1c See the documentation:
NOTE: Minutes-period is for recurring wakeups. E.g. when set at 5 it will wakeup at 0, 5, 10, 15, ..., 55 past the hour. The largest meaningful value is 30
2. The piJuice watchdog timer is reset every time you send a I2C command to the PiJuice. The purpose is to detect a software hang on the Pi.
`sudo reboot' not working is because the 5V to the Pi never goes off.
Pressing the button forces a short 5V interruption, hence the Pi boots.
3. Wakeup on charge has to be enabled at the latest at the previous shutdown.
!.5 firmware and later have a possibility to store the setting in non-volatile memory.
See the changelog for the 1.5 firmware:
https://github.com/PiSupply/PiJuice/blob/master/Software/Source/debian-base/changelog

bjohas, 2023-07-19:
Hi @tvoverbeek , that's really very helpful, thank you so much.

> `sudo reboot' not working is because the 5V to the Pi never goes off.

The shutdown timer has a function to cut power after 60 seconds. Is there a way of sending a "cut power" command prior to sudo reboot?

bjohas, 2023-07-21:
Btw. I should add that some of the devices we've set up do allow a reboot... so I don't fully understand why it's intermittent. However, a reliable way of making re-booting work would be very helpful. I assume there's a way to programme the PiJuice to cut off power and then turn back on?