PiSupply/PiJuice #996: When Pijuice runs out of battery

jorgrde1, 2023-07-24:
Hello!
I'm working with Pijucie and their solar panels, I've configured the Pijuice settings so when it arrives at 15% of battery it shut downs the Rpi and turns on again when reaches the 25%, but I have noticed that it reaches 0% and it does not turn on anymore.
Could be possible that if the pijuice reaches 0% do pijuice orders reset?

tvoverbeek, 2023-07-24:
How do you shut down the Rpi at 15%?
If you use e.g. SYS_FUNC_HALT the Rpi shuts down, but the Pijuice keeps supplying 5V to the Rpi.
The Rpi GPU is still on and also peripherals will continue using the 5V supply -> battery reaches 0%.
Use SYS_FUNC_HALT_POW_OFF instead which will cut the 5V to the Ppi after ca 30 seconds.
Then the PiJuice goes to low power mode (slowly blinking very weak LEDs).

jorgrde1, 2023-07-24:
I use SYS_FUNC_HALT_POW_OFF, but I dont know why it seems it doesnt work all the time, I monitor the information via mqtt and the battery data regarding time are as follows, in principle it works correctly, it looks like it oscillates between the values ​​25 and 15 because it turns off and on, but then there comes a moment where it does not turn off and drops to 0 and from there it never turns on again despite the fact that it is charging due to the solar panel.
[image]

jorgrde1, 2023-07-24:
Also this is my system task from pijuice_cli
[image]

And system events:
[image]

OhSoGood, 2023-07-31:
Thank @bjohas for sharing your scripts