PiSupply/PiJuice #998: pijuice_cli triggers ValueError: 'UNKNOWN' is not in list when changing buttons behaviour

MarcoMobilio, 2023-07-27:
Hi everyone, 

I just installed and updated the PiJuice CLI and firmware, versions should be 1.6 and V1.6_2021_09_10.

whenever I try to assign a function to a button when it was not set before (shows default 'UNKNOWN' as a function in the list for such button/action) it throws an error and crashes the tool. the error is:

```
...
File "/usr/bin/pijuice_cli.py", line 776, in _set_function
self.bgroup[self.FUNCTIONS.index(self.current_config[sw_id][action]['function'])].toggle_state()
ValueError: 'UNKNOWN' is not in list

```

I am sorry if I cannot post the entire exception, but I am in an environment where I cannot connect via SSH so I can only access directly the Pi (which does not have a GUI).

I think it could be that the default value of 'UNKNOWN' is looked for in the dict, but is not present there and throws the error. Maybe there's an issue in the default value or in the handling of the UNKNOWN case?

tvoverbeek, 2023-07-27:
????
There is no 'UNKOWN' default setting for the buttons. It should report 'NO_FUNC' if no function is assigned yet.
The only place in pijuice_cli where 'UNKNOWN' appears is in the firmware update reporting an unknown error.
Are you sure the firmware updating went OK?
What does the firmware tab show in pijuice_cli?
Mine shows this:
[image]
How did you do the update?

MarcoMobilio, 2023-07-28:
Hi, 
I updated using the PiJuice CLI, 

the cli reported UNKNOWN for every possible interaction with SW2 and SW3, I ended up doing a dump of the config file using the pijuice_util.py (you can se a picture of the screen, again, sorry, but its under a very limited network) and there are clearly the UNKNOWN values for each function.

However, if you use the same util to load back the configuration file in the pijuice_cli, it fixes it and now it shows NO_FUNC for the unassigned functions and does not crash any more.

tvoverbeek, 2023-07-28:
OK. It seems to work now for you.
I still cannot understand/explain how you can get the UNKNOWN function names.
The button settings are internal to the firmware and not part of the pijuice service data (/var/lib/pijuice/pijuice_config.JSON)
How did you get the dump showing on your screenshot? With pijuice_util.py?

MarcoMobilio, 2023-07-30:
yes, I used the pyjuice_util, my first idea was to use it to dump a valid config file, substitute UNKNOWN wit my desired functions and load the config again with the same tool, but noticed that upon load also the functions I did not modify became NO_FUNC, which does not trigger the crash anymore. I am still having issues in actually using the buttons, but I don't think that problem is related to this situation.

tvoverbeek, 2023-07-30:
Question still remains how did you get the config file with the UNKNOWN values?
It is **not** part of the install and the JSON file in /var/lib/pijuice/pijuice_config.JSON is automatically updated by the pijuice service.

MarcoMobilio, 2023-07-31:
that I don't know, I just followed the readme and installed pijuice-base with apt.