PiSupply/PiJuice #999: PiJuice stopped charging after outage

markdepalma, 2023-08-05:
I had been using my PiJuice in a setup to monitor a travel trailer. It was working fine and the last time I lost power for an extended amount of time it completely stopped charging. I bought a new battery thinking it may be a battery issue. After putting this battery in it now shows 20% and battery present (was showing 0% and no battery), but it just stays at 20%. It will run off battery power now, but will not charge when plugged in. Not sure what to check.

tvoverbeek, 2023-08-05:
First thing to check is charging enabled?
(https://github.com/PiSupply/PiJuice/tree/master/Software#pijuice-hat-general-config-menu for the GUI or https://github.com/PiSupply/PiJuice/tree/master/Software#general for the CLI)
Where is the external power connected to the Pi or the PiJuice?
If to the Pi also check if GPIO input is enabled.

markdepalma, 2023-08-05:
Went through software one more time and and found that charging was disabled. Is there any reason why this would turn off on its own? It was definitely enabled before the issue and had gone through multiple charge cycles.

tvoverbeek, 2023-08-05:
If the PiJuice loses power and the MCU goes off the 'charging enabled' setting goes off unless it is stored in non-volatile memory. The default is to **not** store it in non-volatile memory.
You can check with this python code:
```
from pijuice import PiJuice
pj=PiJuice(1, 0x14)
pj.config.GetChargingConfig()
```
My output is:
`{'data': {'charging_enabled': True}, 'non_volatile': False, 'error': 'NO_ERROR'}`
If my PiJuice completely loses power (takes more than a week) then the setting is gone.
The CLI/GUI sets the non-volatile attribute to True only if you change the setting.
If you never changed the setting the non-volatile attribute is False

markdepalma, 2023-08-05:
> If the PiJuice loses power and the MCU goes off the 'charging enabled' setting goes off unless it is stored in non-volatile memory. The default is to **not** store it in non-volatile memory. You can [...]

Thank you for this! I just checked now it the setting on and it shows non-volatile:

```
from pijuice import PiJuice
pj=PiJuice(1, 0x14)
pj.config.GetChargingConfig()

{'data': {'charging_enabled': True}, 'non_volatile': True, 'error': 'NO_ERROR'}
```
So, I just checked and this changed to a non volatile setting. Did this change just by flipping that switch in the CLI? How would it have been charging before without that switch checked? What state would the switch have been in before to be enabled, but not enabled as a non-volatile setting?