PiSupply/PiJuice #1000: Pi Juice Failed to execute on new install

krm240, 2023-08-12:
Seems that the bug has not been fixed.  I have the gui running great on 1 Rpi4 running pythin3, but a new install, and I get the Failed to execute child process"/usr/bin/pijuice_gui (no such file or directory).   Any suggestions??

krm240, 2023-08-12:
PiJuice icon is on the top right , shows battery % when hover, but settings does not work, but does not produce error.   Command line start does not work either.  Software version 1.8, firmware 1.3

krm240, 2023-08-12:
Command line gives the folloing error: 
bash: /usr/bin/pijuice_cli: No such file or directory

krm240, 2023-08-12:
Running python 3.9.2
Hope this all helps...

krm240, 2023-08-12:
running 
sudo -u pijuice python3 /usr/bin/pijuice_gui.py
brings up the gui....
Is it a permission issue?

tvoverbeek, 2023-08-12:
Probably somethin went wrong when you did the new install.
/usr/bin/pijuice_gui and /usr/bin/pijuice_cli are setuid executables (uid pijuice) which call the python3 interpreter to run /usr/bin/pijuice_gui.py or /usr/bin/pijuice_cli.py.
This is eetup when you install the pijuice-base package.
Can you list the output from `ls -l /usr/bin/pijuice*`?

krm240, 2023-08-12:
***@***.***:~ $ ls -l /usr/bin/pijuice*?

lrwxrwxrwx 1 root    root        22 Aug 11 21:01 /usr/bin/pijuiceboot -> /usr/bin/pijuiceboot64

-rwxr-xr-x 1 root    root     22732 Jan 29  2022 /usr/bin/pijuiceboot32

-rwxr-xr-x 1 root    root     23904 Jan 30  2022 /usr/bin/pijuiceboot64

lrwxrwxrwx 1 root    root        22 Aug 11 21:01 /usr/bin/pijuice_cli -> /usr/bin/pijuice_cli64

-rwsr-xr-x 1 pijuice pijuice   9096 Dec 17  2018 /usr/bin/pijuice_cli32

-rwsr-xr-x 1 pijuice pijuice  14872 Jan 30  2022 /usr/bin/pijuice_cli64

-rw-r--r-- 1 pijuice pijuice 101552 May 16  2021 /usr/bin/pijuice_cli.py

lrwxrwxrwx 1 root    root        22 Aug 11 21:01 /usr/bin/pijuice_gui -> /usr/bin/pijuice_gui64

-rwsr-xr-x 1 pijuice pijuice   9096 Dec 17  2018 /usr/bin/pijuice_gui32

-rwsr-xr-x 1 pijuice pijuice  14872 Jan 30  2022 /usr/bin/pijuice_gui64

-rw-r--r-- 1 pijuice pijuice 105446 May 16  2021 /usr/bin/pijuice_gui.py

-rwxr-xr-x 1 root    root     11275 Jan 30  2022 /usr/bin/pijuice_log.py

-rwxr-xr-x 1 root    root     15664 Jan 30  2022 /usr/bin/pijuice_sys.py

-rwxr-xr-x 1 root    root      6205 Jan 30  2022 /usr/bin/pijuice_tray.py

***@***.***:~ $ ^C

tvoverbeek, 2023-08-13:
Everything looks OK.
From your listing I gather you are running a 64-bit version of the OS. Correct?
Here is the same listing from my 32-bit system:
```
tvo@rpi32-deb11:~ $ ls -l /usr/bin/pijuice* --color=none
lrwxrwxrwx 1 root    root        22 Feb  9  2022 /usr/bin/pijuiceboot -> /usr/bin/pijuiceboot32
-rwxr-xr-x 1 root    root     22732 Jan 29  2022 /usr/bin/pijuiceboot32
-rwxr-xr-x 1 root    root     23904 Jan 30  2022 /usr/bin/pijuiceboot64
lrwxrwxrwx 1 root    root        22 Feb  9  2022 /usr/bin/pijuice_cli -> /usr/bin/pijuice_cli32
-rwsr-xr-x 1 pijuice pijuice   9096 Dec 17  2018 /usr/bin/pijuice_cli32
-rwsr-xr-x 1 pijuice pijuice  14872 Jan 30  2022 /usr/bin/pijuice_cli64
-rw-r--r-- 1 pijuice pijuice 101552 May 16  2021 /usr/bin/pijuice_cli.py
lrwxrwxrwx 1 root    root        22 Feb  9  2022 /usr/bin/pijuice_gui -> /usr/bin/pijuice_gui32
-rwsr-xr-x 1 pijuice pijuice   9096 Dec 17  2018 /usr/bin/pijuice_gui32
-rwsr-xr-x 1 pijuice pijuice  14872 Jan 30  2022 /usr/bin/pijuice_gui64
-rw-r--r-- 1 pijuice pijuice 105446 May 16  2021 /usr/bin/pijuice_gui.py
-rwxr-xr-x 1 root    root     11275 Jan 30  2022 /usr/bin/pijuice_log.py
-rwxr-xr-x 1 root    root     15664 Jan 30  2022 /usr/bin/pijuice_sys.py
-rwxr-xr-x 1 root    root      6205 Jan 30  2022 /usr/bin/pijuice_tray.py
tvo@rpi32-deb11:~ $ 
```
/usr/bin/pijuice_gui is symbolic link to either the 32-bit or 64-bit version.
/usr/bin/pijuice_gui64 has the setuid bit set and has owner pijuice.
I do not know why it does not work.
Can it be that you do not allow setuid executables?

krm240, 2023-08-13:
I thought I installed 32 bit os version

If 64 bit os is an issue, I can resinstall 32, I am ok with that

Is there a way to tell if 32 or 64?

 

***@***.***:~ $ cat /etc/os-release

PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"

NAME="Raspbian GNU/Linux"

VERSION_ID="11"

VERSION="11 (bullseye)"

VERSION_CODENAME=bullseye

ID=raspbian

ID_LIKE=debian

HOME_URL=http://www.raspbian.org/

SUPPORT_URL=http://www.raspbian.org/RaspbianForums

BUG_REPORT_URL=http://www.raspbian.org/RaspbianBugs

 

Ken

krm240, 2023-08-13:
Can it be that you do not allow setuid executables?

Not sure how to do that  - so if I did stop it, not intentional…

Os says 32

***@***.***:~ $ getconf LONG_BIT

32

tvoverbeek, 2023-08-13:
Since the move to kernel 6.1 installing raspberry pi os on a pi4 defaults to using a 64-bit kernel for 32-bit user programs.
This breaks the detection of 32-bit vs 64-bit in the pijuice-base install procedure which relies on the output of `uname -a`.
On a pi4 with default os install it will report aarch64.
See e.g. https://raspberrypi.stackexchange.com/questions/143287/i-installed-raspbian-32-bit-but-arch-writes-aarch64

To get your recent pi4 install working, manually change the symbolic links:
```
$ sudo rm /usr/bin/pijuice_gui
$ sudo ln -s  /usr/bin/pijuice_gui32 /usr/bin/pijuice_gui
```
Same for /usr/bin/pijuice_cli and /usr/bin/pijuice_boot

krm240, 2023-08-13:
Will it work without adjustments if I install 64bit  OS on the Pi?

Ie: I believe that I can use 32 or 64, no diff to me that I know of.

So if I go 64, will that resolve the issues?

Ken

krm240, 2023-08-13:
Did as requested, and it now works

Thanks

Ken

krm240, 2023-08-13:
Ok, here is another issue:

 

I thought the issue was because of the previous issue.

So just now tried to update the firmware and received this issue.

Any thoughts?

 

Ken

tvoverbeek, 2023-08-13:
Firmware update in the cli/gui uses pijuiceboot internallly. So you have to change the pijuiceboot symbolic link to /usr;/bin/pijuiceboot32

From my answer above:
**Same for /usr/bin/pijuice_cli and /usr/bin/pijuice_boot**

Bonzadog, 2023-10-08:
> $ sudo ln -s  /usr/bin/pijuice_gui32 /usr/bin/pijuice_gui

Did not work .
nt@raspberrypi:~ $ ls /usr/bin/pijuice_boot
ls: cannot access '/usr/bin/pijuice_boot': No such file or directory

I assume that this will be soon corrected.  Very annoying when installation procedures fail.
I hope this will be corrected and released.

quitmeyer, 2023-11-20:
> $ sudo rm /usr/bin/pijuice_gui $ sudo ln -s /usr/bin/pijuice_gui32 /usr/bin/pijuice_gui

This advice worked for me! Thanks a bunch!

tvoverbeek, 2025-01-26:
This is a well-known problem caused by an error in detecting if you are running on 64-bit or 32-bit.
Especislly on a Pi4 which uses a 64-bit kernel but a 32-bit user-land.
Run the following in a terminal window:
```
sudo rm /usr/bin/pijuiceboot 
sudo ln -s /usr/bin/pijuiceboot64 /usr/bin/pijuiceboot
sudo rm /usr/bin/pijuice_cli
sudo ln -s /usr/bin/pijuice_cli64 /usr/bin/pijuice_cli
sudo rm /usr/bin/pijuice_gui
sudo ln -s /usr/bin/pijuice_gui64 /usr/bin/pijuice_gui
```