PiSupply/PiJuice #1002: Wakeup on charge not saving correctly with pijuice_cli

markdepalma, 2023-08-20:
When trying to set "Wakeup on charge" on via pijuice_cli it doesn't properly commit. I was having issues getting the feature to work, so I checked the value with Python and it wasn't turned set. I set the value and set it as non_volitile in Python and after doing this the feature worked, but pijuice_cli still does not recognize the setting. Not having this issue with other settings.

Running pijuice-base/stable 1.8.

tvoverbeek, 2023-08-21:
@markdepalma Thanks for reporting this.
I can  confirm that pijuice_cli does not set the trigger level value.
pijuice_gui is even worse, it reports the value as set (but takes it form the config file), but never updates the value in the PiJuice.
Working on a proper solution.

Wahaj6Ahmad, 2023-12-12:
Is there a workaround to this problem?
I am trying to set the wakeup on charge dynamically through the python code using the command SetWakeUpOnCharge().
The script checks the current charge level and changes the wakeup on charge value accordingly. The python script runs without errors and GetWakeUpOnCharge() returns the desired output. However, this change is not reflected in the config file or the pijiuce-gui. Furthermore, the pi does not wake up on the set charge level either.

markdepalma, 2023-12-12:
> Is there a workaround to this problem? I am trying to set the wakeup on charge dynamically through the python code using the command SetWakeUpOnCharge(). The script checks the current charge level [...]

I was able to use Python to get/set these values without issue. These are my snippets:
`#PiJuice Python Get Charging Config
python
from pijuice import PiJuice
pj=PiJuice(1, 0x14)
pj.config.GetChargingConfig()

#PiJuice Python Get Wakeup on Charge
python
from pijuice import PiJuice
pj=PiJuice(1, 0x14)
pj.power.GetWakeUpOnCharge()

#PiJuice Python Set Wakeup on Charge
python
from pijuice import PiJuice
pj=PiJuice(1, 0x14)
pj.power.SetWakeUpOnCharge(20, True)`

tvoverbeek, 2023-12-12:
GetWakeUpOnCharge always returns the non-volatile setting.
@markdepalma It works for you since you set the non-volatile value (The True argument in SetWakeUpOnCharge.

Wahaj6Ahmad, 2023-12-13:
I am also setting the non-volatile value to be true. I am attaching my code with this comment for better understanding.
The idea is to read the current charge level and set the WakeUpOnCharge value to be 20% more. The actual script also sets the wake-up alarm but I removed it from the code for brevity. I have also attached the console output of the script.
Note: After executing this script, the result is not reflected in the pijuice-gui or the config file. The pi does not wake up either. I have tested this at various charge levels.

```
#!/usr/bin/python3

from pijuice import PiJuice # Import pijuice module

import subprocess

pijuice = PiJuice(1, 0x14) # Instantiate PiJuice interface object

print('PiJuice Status::', pijuice.status.GetStatus()) # Read PiJuice status.

def pi_pwr_off_shutdown(delay): #Removes power from the pj to the pi. delay is given to do a safe shutdown. min 30 sec recommended

	pijuice.power.SetPowerOff(delay) 

	print("PiPwrOffShutdown::", pijuice.power.GetPowerOff())

	# Run the shutdown command

	subprocess.call("sudo shutdown -h now", shell=True) #shuts down the raspberry pi

def set_wakeup_on_charge(): #Set wake up on charge dynamically 20% more than the current charge

	

	#Get current charge level of the battery

	currCharge = int(pijuice.status.GetChargeLevel()['data'])

	print('SetWakeupOnCharge:: Current Charge Level:', currCharge)

	

	if currCharge <= 75: setCharge = currCharge + 20 #The wakeup charge level will be 20% more than the current level.

	else: setCharge = 95 #and 95% for charge more than 75

	pijuice.power.SetWakeUpOnCharge(setCharge, True)

	status = pijuice.power.GetWakeUpOnCharge()

	print("SetWakeupOnCharge:: Wake up on charge status:", status)

	if not status['data'] == 'DISABLED':

		return True

	return

if __name__ == "__main__":

	wakeup_on_charge_status = False

	while wakeup_on_charge_status != True:

		wakeup_on_charge_status = set_wakeup_on_charge()

	pi_pwr_off_shutdown(30)
```

[image]

tvoverbeek, 2023-12-13:
In order for wakeuponcharge to work the 5V power to the PiJuice has to be off.
In other words, you have to be running on battery power.
After the 30 seconds the power led of the Pi has to be off and the PiJuice Leds have to blink very slowly and faint.
Then you can restore the external power to the PiJuice, so the battery is charging again.
If you leave the external power to the PiJuice on, the WakeUpOnCharge is cancelled by the firmware

Wahaj6Ahmad, 2023-12-14:
This is not very handy, as my system is remote and solar-powered. Can we cut off 5V power to PiJuice through software?
Also, it is worth mentioning that the WakeUpOnCharge works fine when set through the GUI.