PiSupply/PiJuice #1003: Python3-urwid preventing install?

MattGeale, 2023-08-23:
Hi All,

Hoping someone could give me a bit of a hand with this error.

I'm trying to initialise the PiJuice hat on a ras pi zero 2w and keep getting an error surrounding urwid saying it can't be found doing both ```sudo apt-get install pijuice-base``` (preference) or ```sudo apt-get install pijuice-gui```.

I've googled python3-urwid and whenever I try and install it using the [commands I've found](https://github.com/urwid/urwid/wiki/Installation-instructions), i.e. ```sudo apt-get install python-urwid```, it says the package cannot be found...

[image]

Does anyone have any ideas what I'm missing or whether there are commands I'm meant to do pre running the install pijuice-base command?

Thanks in advance :)

tvoverbeek, 2023-08-23:
First`sudo apt-get install python-urwid` installs the python2 version of the package.
Raspberry Pi OS now only provides python3 not python2.
If you look at the description of the python3-urwid package:
``` 
tvo@rpi32-deb11:~ $ apt show python3-urwid
Package: python3-urwid
Version: 2.1.2-1
Priority: optional
Section: python
Source: urwid
Maintainer: Debian Python Team <team+python@tracker.debian.org>
Installed-Size: 967 kB
Provides: python3.9-urwid
Depends: python3 (<< 3.10), python3 (>= 3.9~), python3:any, libc6 (>= 2.4)
         ------------------------------------
,,,
```
It seems the pyrhon3-urwid package requires only python 3.9.x
If you are using python 3.10 the python3-urwid package will not install.

Also the error message says `you have held broken packages`.
You can list held packages with `apt-mark --showhold`

MattGeale, 2023-08-24:
I can't even get it to find python3-urwid...
[image]

Tried as you suggested the apt-mark, and just returned this:
[image]

closed 2023-08-25 by MattGeale