PiSupply/PiJuice #1005: COMMUNICATION_ERROR with Model 3B+

nmoorcroft, 2023-08-29:
I have installed a brand new PiJuice HAT on my Model 3B+ and am getting a COMMUNICATION_ERROR when I look at the status in pijuice_cli. I have enabled the I2C interface but i2cdetect shows nothing is connected.

`pi@homebridge:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- `

Is the HAT broken? Should I be returning it for a replacement?

tvoverbeek, 2023-08-29:
Do the i2c device (/dev/i2c-1) exist?
What is the output from `ls -l /dev/i2c*`?
Does `sudo i2cdetect -y 1` produce different output than `i2cdetect -y 1`?
Are you sure the 40-pin header is correctly installed (e.g. not shifted by one row)?

nmoorcroft, 2023-08-30:
Hi, thanks for helping. The output from `ls -l /dev/i2c*` is
```
pi@homebridge:~ $ ls -l /dev/i2c*
crw-rw---- 1 root i2c 89, 1 Aug 29 17:38 /dev/i2c-1
```
`sudo i2cdetect -y 1` produces the same output as `i2cdetect -y 1`
```
pi@homebridge:~ $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --   
```
The 40-pin header is correctly installed and the HAT is screwed to the Pi.

tvoverbeek, 2023-08-30:
How are you powering the 3B+/PiJuice combination?
If powering via the 3B+ try to power via the PiJuice micro-USB.
Any life on the PiJuice LEDs?
If the PiJuice LEDs stay off then obviously the PiJuice MCU is not powered and hence cannot reply to I2C transactions.

nmoorcroft, 2023-08-30:
I have been powering via the 3B+ micro-USB, if I use the PiJuice micro-USB the RPI doesn't power up.
Non of the PiJuice LEDs are on. If I remove the battery and reconnect it, LED lights up green for a few ms then goes out.

tvoverbeek, 2023-08-30:
This smells like 5V from Pi to PiJuive via GPIO is blocked and the battery is completely empty.
Try the following: connect external power to the PiJuice with battery inserted and check if LED1 blinks (alternating blue/yellor or red means charging battery level < 50%, alternating blue/green means charging battery level > 50%).
If so, momentarily press SW1. This should start the Pi.

nmoorcroft, 2023-08-30:
I've connected the external power to the PiJuice with the battery inserted, unfortunately non of the LEDs blink and SW1 doesn't do anything.

tvoverbeek, 2023-08-30:
Then I am out of suggestions. It surely looks like it is broken. Return it and ask for a replacement.

nmoorcroft, 2023-08-30:
Ok. Thanks for your help.

nmoorcroft, 2023-09-10:
Received replacement. Working as expected now, thanks.

closed 2023-09-10 by nmoorcroft