PiSupply/PiJuice #1006: Stacking pHats on PiJuice Zero / suitable case for batteries + one pHat

r-colvin, 2023-08-31:
Hi!
 
I am building a project and I need the ability to schedule power on / off … say 10 minutes on starting 5 minutes before the hour and ending 5 minutes past the hour. The point being to run a PiZero for as long as possible in a location (my shed!) where I don’t have mains power. I can probably attach a USB Power Bank to extend the time – but still having the schedule ability will be required to get weeks of constant run time minimum.
 
For a battery, I am considering if a flat LiPo is the better option or an 18650 in a holder which I can then easily swap in and out if need be.
 
However, I want to have the PiZero, the Pi Juice (with a battery) and then add a pHat – specifically this one: https://thepihut.com/products/nb-iot-emtc-edge-gprs-gnss-hat-for-raspberry-pi?variant=39808804815043 – but I can’t work out if there will be a pin conflict; I plan on using the HAT to provide a ppp connection to the Pi for internet access (my wifi doesn’t stretch to the shed either)
 
If I can stack, then I am trying to work out the best way to package everything; using a flat lipo, the PiJuice Zero case looks good – but I don’t think it will hold the third pHat! So this one caught my eye : https://thepihut.com/products/flanged-weatherproof-enclosure-with-pg-7-cable-glands?variant=18634584817726 :  though might not be tall enough? Also I am worried about how to add a battery safely to avoid header pins and solder joints …. So any recommendations would be welcome!

tvoverbeek, 2023-08-31:
Regarding the statement "Turning on the Raspberry Pi via the onboard intelligent switch only works when the power is provided to the micro USB on the PiJuice.", What you want is a scheduled wakeup at 5 minutes before the hour using the PiJuice RTC. If  the battery has sufficient charge you do not need to provide external power to the PiJuice.
You will also need to get the GSM antenna cable out of the case.
I cannot tell if your case (the PiHut) one is high enough for Pi Zero + GSM modem + PiJuice.
If you use shutdown with 5V off (to save battery charge) then the GSM modem has to log in every time the Pi goes on.
GSM modems can be quite power hungry. 
Anyway your PiHut case is also available on Adafruit: https://www.adafruit.com/product/3931 with even more background info.
Hope this helps (a littel bit).

r-colvin, 2023-08-31:
hey @tvoverbeek thanks for the update - appreciate it :)

reading your answer it seems that I can use the PiJuice Zero to power my Pi Zero from battery directly through the GPIO header and create a script that will automatically power on and off the Pi on a schedule - this is perfect! 

however... reading https://github.com/PiSupply/PiJuice I am still confused
- in https://github.com/PiSupply/PiJuice#power-on-and-off-the-pi it says
"If you wish to start your Raspberry Pi using the PiJuice SW1 you'll have to connect the micro USB to the PiJuice directly."
and then a little further it seems to contradict this, by saying:
"To power on the Raspberry Pi single press SW1 briefly as you would for a click of a mouse. If the micro USB is directly connected to the Raspberry Pi then it will power up immediately without you needing to press the SW1."

... oh, I think I see my confusion (and this would relate also to "Turning on the Raspberry Pi via the onboard intelligent switch only works when the power is provided to the micro USB on the PiJuice." - the PiJuice has a micro USB port for power-in, as does the Pi Zero - and you can use either... (I am used to devices where you power via a Hat/pHat, that you shouldn't use both) - I had interpreted these statements as saying you needed to run a micro USB between the PiJuice and the Pi Zero! 

Right - so, in my case, I will have the Pi Zero stacked on the PiJuice Zero - and I will have a LiPo attached by JST (the biggest I can find); I may, use a PowerBank as a supplementary power source; is there a difference in behaviour if I plug the PowerBank into the Pi Zero or PiJuice? Is there a preference for my use case? Essentially I want to keep the PiJuice battery high enough to keep the RTC running to bring the Pi Zero up and down. Makes sense...

Looking at the docs for the case, I am hopeful everything will fit, and if not... the case is not so expensive. I have a SMA cable coming to run the antenna from - and that will run through the cable duct, so no issue there.

Regarding the modem, I am speccing an LTEm modem for this exact reason where I can use PSM and eDRX to reduce the power consumption when re-connecting (which is what these functions were designed for)

looks like I might have a plan.. thanks again

tvoverbeek, 2023-09-01:
Good luck with your project.
Closing this issue. Feel free to open a new one (or a discussion) if you have more questions.

closed 2023-09-01 by tvoverbeek