PiSupply/PiJuice #1008: New User: Confusion of missing pogo pin and information regarding DIP switch settings

EnnPeeTee, 2023-09-04:
I am reposting the content of an email sent to the company, because apparently all technical enquiries are managed here.

Greetings!
 
I wonder if you can please advise me?
 
I have just received a PiJuice for use with a Pi 3B+.   I must say it looks stonkingly exciting and nicely designed.
 
After the headache faded from trying to read the enclosed ‘manual’,  I trawled the internet and eventually stumbled across some operational installation instructions at
https://learn.pi-supply.com/make/pijuice-quick-start-guide-faq/#software-installation
 
•	It refers to a pogo-pin which is not included.   This is also referred to at https://uk.pi-supply.com/collections/pijuice/products/pijuice-standard.  Do I need a pogo pin or has it been designed out now?
 
•	There is also another area of concern, the small DIP switch settings. It tells me it is important to set them in a specific way, yet the dip switch on my device is covered with a film which suggests ‘leave this alone!’ Mine are both in the central position, unlike the image on the web-page.  What should I do?

tvoverbeek, 2023-09-04:
Pogo-pin is not needed any more since 2019.
From the hardware sectionin the github hardware README: https://github.com/PiSupply/PiJuice/tree/master/Hardware#unpopulated
```
=========================================================================
Note: As of Firmware v1.3_2019_01_15 the pogo pin is no longer required to wakeup the
Raspberry Pi  when powering from the Pi micro USB. A single press of SW1 on the PiJuice will
wakeup the Pi or a short of  GPIO pins 5 & 6 will also wakeup the Pi. This method has been
introduced to accommodate the new Raspberry Pi boards and any future boards so that
hardware modifications are not needed to accommodate new run pin locations.
Pin 6 is ground and pin 5 is GPIO3. When the Pi is shut down (in low power state) a falling
edge on GPIO3 (e.g. shorting pins 5 and 6) will make the GPU firmware boot the Pi.
Note: this implies the Pi still has 5V power. Since the ARM is not active yet it does not
interfere with I2C which is configured during the ARM boot. So there is nothing to do
to check if things are properly configured. The watchdog reset uses the same sequence
as the other wakeups (interrupt 5V when powered via the PiJuice, otherwise use run pin
and finally if no run pin use the GPIO3 method).
=========================================================================
``` 
The DIP swiches are only usefull if you want to use the PiJuice HAT without RPi software support.
It provides a (partial) choice of battery profiles.
See https://github.com/PiSupply/PiJuice/tree/master/Hardware#dip-switches-hat-only
If you use the RPi software support (pijuice_gui/pijuice_cli) there is no need for the disp switches.

Read both the software and hardware README on github for much more information.

EnnPeeTee, 2023-09-04:
Brilliant! Thank you for such an incredibly speedy response! Extremely helpful.

closed 2023-09-07 by EnnPeeTee