PiSupply/PiJuice #1011: Battery NON_PRESENT

nmoorcroft, 2023-09-09:
Under HAT status the battery is showing as NOT_PRESENT

[image]

i2cdetect is showing the HAT is connected

```
pi@homebridge:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --  
```

LED1 is blinking red.

Any ideas?

nmoorcroft, 2023-09-10:
Solved. Left RPI plugged in overnight with HAT connected. Disconnected from power and removed the battery in the morning. Replaced the battery and powered on with SW1, without reconnecting the power and the RPI booted. Connected power to the HAT and it's now showing as CHARGING_FROM_IN.

closed 2023-09-10 by nmoorcroft

martinarisk, 2023-11-05:
Thank you. Initially battery is NOT_PRESENT and close to 0V. Solved by:

1. Uncomment dtparam=i2c_arm=on in /boot/config.txt
2. Setting the DIP switch according to my battery. 
3. Remove the film on the battery and plug battery in.
4. Left RPI sit a while blinking the blue+red led while connected to the hat only via micro usb.
5. Start it using sw 1 button while disconnected from power. RPI board should turn on green light.
6. Connect power to usb c only. Can now close the case.
7. Notice battery in sudo pijuice_cli is now CHARGING_FROM_5V_IO. Let it charge to 50%
8. Update firmware to 1.6 from 1.5 (optional) from the pijuice_cli firmware menu.