PiSupply/PiJuice #1012: PiJuice DATA_CORRUPTED

rithwikmakesthings, 2023-09-24:
Hi, I have a Pi4B with Bullseye installed, no other I2C devices and I have encountered the DATA_CORRUPTED error. I have tried modifying config.txt as suggested but that didn't help. I've also hard reset PiJuice by holding SW1 and SW2 for 20 seconds, but that hasn't done anything either.
I've run i2cdetect -y 1, and that returns all slave devices (output pasted below)
Any help is greatly appreciated!

pi@pi:~ $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         08 09 0a 0b 0c 0d 0e 0f 
10: 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 
20: 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 
30: 30 31 32 33 34 35 36 37 38 39 3a 3b 3c 3d 3e 3f 
40: 40 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f 
50: 50 51 52 53 54 55 56 57 58 59 5a 5b 5c 5d 5e 5f 
60: 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 
70: 70 71 72 73 74 75 76 77

tvoverbeek, 2023-09-24:
This is typical when there is ahort to ground on SDA/GPIO 2/Pin 3.
Also if you have another program setting GPIO 2 as output the same will happen.
i2cdetect works by sending the address (via SDA), changing SDA to input (with pull-up) and checking if the addressed unit pulls SDA low.
Also are you sure the PiJuice is properly installed on the 49-pin header? Typical problems are connections shifted by 1 row and/or column.
Another check is to run i2cdetect without any unit connected to the i2c bus. There should be no unit responding.

rithwikmakesthings, 2023-09-25:
Thank you for the response!
I'm 100% sure it is installed properly, the issue occurs when I use the PiJuice unit connected via jumper cables as well.
I've also tested it on a fresh install of Raspbian with only the PiJuice application installed, so it is unlikely that there is interference from other programs...
Running i2c detect without any unit gives the expected blank response.

I've also tested my pi with another i2c device, and that one seems to be performing as expected... Do you reckon I'm looking at a busted PiJuice? If so, is this something that can be fixed? 

Thanks again!