PiSupply/PiJuice #1014: -bash: /usr/bin/pijuice_cli: No such file or directory

ECHEngineering, 2023-10-04:
I've noted a few older threads on this...those fixes either did not quite match or the fixes did not work for me.

I've installed a PiJuice onto a Pi 3b.  The Pi is actually an EmonBase so I did not set it up myself.  I've connected as user pi (suggested in the instructions) with putty.  I then ran the installer sudo apt-get install pijuice-base.  It seemed to work okay.
Only having the command line access to the pi through putty, I was hoping to setup the pijuice using pijuice_cli.   When I enter pijuice_cli I get the error no such file.   When I enter pijuice_cli.py I get:

pi@emonhp:~ $ pijuice_cli.py
/usr/bin/pijuice_cli.py: line 6: import: command not found
/usr/bin/pijuice_cli.py: line 7: import: command not found
/usr/bin/pijuice_cli.py: line 8: import: command not found
/usr/bin/pijuice_cli.py: line 9: import: command not found
/usr/bin/pijuice_cli.py: line 10: import: command not found
/usr/bin/pijuice_cli.py: line 11: import: command not found
/usr/bin/pijuice_cli.py: line 12: import: command not found
/usr/bin/pijuice_cli.py: line 13: import: command not found
/usr/bin/pijuice_cli.py: line 15: import: command not found
/usr/bin/pijuice_cli.py: line 16: from: command not found
/usr/bin/pijuice_cli.py: line 18: BUS: command not found
/usr/bin/pijuice_cli.py: line 19: ADDRESS: command not found
/usr/bin/pijuice_cli.py: line 20: PID_FILE: command not found
/usr/bin/pijuice_cli.py: line 21: LOCK_FILE: command not found
/usr/bin/pijuice_cli.py: line 26: pijuice: command not found
/usr/bin/pijuice_cli.py: line 28: pijuiceConfigData: command not found
/usr/bin/pijuice_cli.py: line 29: PiJuiceConfigDataPath: command not found
/usr/bin/pijuice_cli.py: line 53: from: command not found
/usr/bin/pijuice_cli.py: line 54: from: command not found
/usr/bin/pijuice_cli.py: line 57: syntax error near unexpected token `('
/usr/bin/pijuice_cli.py: line 57: `def _InitPiJuiceInterface():'

Few other attempts based on other information found on github:

pi@emonhp:~ $ pijuice_cli
-bash: /usr/bin/pijuice_cli: No such file or directory
pi@emonhp:~ $ ls -l /usr/bin/pijuice*
lrwxrwxrwx 1 root    root        22 Oct  4 16:42 /usr/bin/pijuiceboot -> /usr/bin/pijuiceboo                            t64
-rwxr-xr-x 1 root    root     22732 Jan 29  2022 /usr/bin/pijuiceboot32
-rwxr-xr-x 1 root    root     23904 Jan 30  2022 /usr/bin/pijuiceboot64
lrwxrwxrwx 1 root    root        22 Oct  4 16:42 /usr/bin/pijuice_cli -> /usr/bin/pijuice_cl                            i64
-rwsr-xr-x 1 pijuice pijuice   9096 Dec 17  2018 /usr/bin/pijuice_cli32
-rwsr-xr-x 1 pijuice pijuice  14872 Jan 30  2022 /usr/bin/pijuice_cli64
-rwxr-xr-x 1 pijuice pijuice 101552 May 16  2021 /usr/bin/pijuice_cli.py
lrwxrwxrwx 1 root    root        22 Oct  4 16:42 /usr/bin/pijuice_gui -> /usr/bin/pijuice_gu                            i64
-rwsr-xr-x 1 pijuice pijuice   9096 Dec 17  2018 /usr/bin/pijuice_gui32
-rwsr-xr-x 1 pijuice pijuice  14872 Jan 30  2022 /usr/bin/pijuice_gui64
-rw-r--r-- 1 pijuice pijuice 105446 May 16  2021 /usr/bin/pijuice_gui.py
-rwxr-xr-x 1 root    root     11275 Jan 30  2022 /usr/bin/pijuice_log.py
-rwxr-xr-x 1 root    root     15664 Jan 30  2022 /usr/bin/pijuice_sys.py
-rwxr-xr-x 1 root    root      6205 Jan 30  2022 /usr/bin/pijuice_tray.py
pi@emonhp:~ $ usr/bin/pijuice_cli
-bash: usr/bin/pijuice_cli: No such file or directory
pi@emonhp:~ $ sudo -u pijuice python3 /usr/bin/pijuice_gui.py
Traceback (most recent call last):
  File "/usr/bin/pijuice_gui.py", line 2151, in <module>
    start_app()
  File "/usr/bin/pijuice_gui.py", line 2095, in start_app
    root = Tk()
  File "/usr/lib/python3.9/tkinter/__init__.py", line 2270, in __init__
    self.tk = _tkinter.create(screenName, baseName, className, interactive, wantobjects, useTk, sync, use)
_tkinter.TclError: no display name and no $DISPLAY environment variable

Any ideas?

tvoverbeek, 2023-10-04:
iF this is on a Pi3B, then since some time the Linux kernel is running as 64=bit while the user programs are 32-bit.
The detection to either use i.e. the 32-bit or 64-bit version does nit work correctly now on pi3b and pi4 with the 32-bit raspberry pi os.
For a Pi3b delete the 64-bit version of pijuice_cli (/usr/bin/pijuice_cli64), the symbolic link snd rename /usr/bin/pijuice_cli32 to /usr/bin/pijuice_cli. Similar for pijuice_gui and pijuice_boot.
The post_install cript will be modified with the correct test for 32-bit or 64-bit userspace.

ECHEngineering, 2023-10-05:
You sir are a genius.  I would never have got that on my own.   
Thank you so much for your help.
I promised to share my pijuice upgrade of the open energy monitor with the group there, I'll be sure to acknowledge and share your advice too.
Thanks again.

ECHEngineering, 2023-11-12:
The prototype has been running well.  I'm now making more but I can't repeat my initial success...  I'm falling at the first hurdle on this one with the install rather than the renaming the files to get around the 32 / 64 bit issue.  The error implies a Python 3 version is the issue but the Pi and SD card came from the same supplier as the first unit I built.

pi@emonhp:~ $ sudo apt-get install pijuice-base
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 python3-urwid : Depends: python3 (< 3.6) but 3.9.2-3 is to be installed
E: Unable to correct problems, you have held broken packages.
pi@emonhp:~ $ ^C
pi@emonhp:~ $ python3 --version
Python 3.9.2

I did also try sudo apt update and I looked for PiJuice in the usr/bin/ directory.   It is not there.  And I tried to install pijuice-gui and pijuice-cli.   Those ideas met the same end result as above.

tvoverbeek, 2023-11-12:
This is why rhe install fails:
`python3-urwid : Depends: python3 (< 3.6) but 3.9.2-3 is to be installed`
You are trying to install an old pijuice-base packaga.
Current dependency is:
`Depends: python3 (>= 3.5), i2c-tools, python3-smbus, python3-urwid`
it looks like you never ran `sudo aot upgrade`
Dependency on pyrhon3 (>=3.5) has been there since 2018

ECHEngineering, 2023-11-12:
sudo apt update was the command I ran before the text I coped and pasted.
This time I ran 
sudo apt update
then
sudo apt full-upgrade
because there were packages listed that were not updated
Now when I run sudo apt update I get the result all packages are up to date.
Then sudo apt-get install pijuice-base and same error.
.....
pi@emonhp:~ $ sudo apt update
Hit:1 http://archive.raspberrypi.org/debian bullseye InRelease
Hit:2 http://legacy.raspbian.org/raspbian stretch InRelease
Hit:3 https://packages.sury.org/php bullseye InRelease
Hit:4 https://download.docker.com/linux/raspbian bullseye InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
pi@emonhp:~ $ sudo apt-get install pijuice-base
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 python3-urwid : Depends: python3 (< 3.6) but 3.9.2-3 is to be installed
E: Unable to correct problems, you have held broken packages.
....

tvoverbeek, 2023-11-12:
Seems you are on an old debian/raspbian version. What is the output from `cat /etc/debian_version`?
Current version is 12.2 (bookworm).

ECHEngineering, 2023-11-12:
11.7

tvoverbeek, 2023-11-12:
32-bit or 64-bit? Which RPI?
11.7 is buster. If you are running a 32-bit OS on Pi3 and later it will use a 64-bit linux kernel with a 32-bit user-land, The install procedure does not detect this correctly and installs the wrong version of pijuice_cli, pijuice_gui and pijuiceboot.
Can you also provide the output of `ls -l /usr/pin/pijuice*`?

ECHEngineering, 2023-11-12:
ls: cannot access '/usr/pin/pijuice*': No such file or directory

It's a Pi 4.  

I did not set the pi up from scratch unfrotunately so I do not know much about it.  I bought the Pi and SD Cards from the open source energy monitor people.  Same as last time.  It has their software all pre-installed ans set up.
If it helps, your fix on the first one when I could not run the application was to delete the 64 bit version last time and rename the 32 bit version of the application without the 32.  But the install process ran just fine on that one.  So I guess a new problem today   ...sadly I have 35 more to do.  False hope on the first one I guess.  But that was only a month ago and it all came from the same place...  It feels like I must be missing something daft.

tvoverbeek, 2023-11-12:
Typo in my previous comment
Should be: Can you provide the output of `ls -l /usr/bin/pijuice*`? (**bin** instead of **pin**)
In addition the output of `dpkg --print-architecture` and `uname -a`

ECHEngineering, 2023-11-12:
pi@emonhp:~ $ ls -l /usr/bin/pijuice*
ls: cannot access '/usr/bin/pijuice*': No such file or directory
pi@emonhp:~ $ dpkg --print-architecture
armhf
pi@emonhp:~ $ uname -a
Linux emonhp 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
pi@emonhp:~ $

tvoverbeek, 2023-11-12:
OK, armhf means 32-bit user programs. The aarch64 means 64-bit linux kernel.
So thus ia a 32-bit OS.
Both as expected. Did you try to install pijuice-base on this Pi4?
Does `apt-cache search pijuice` produce any output?
Here is my output:
```
tvo@rpi64-deb11:~ $ apt-cache search pijuice
pijuice-base - Basic support for Pi-Supply's PiJuice HAT
pijuice-gui - The desktop applications for the Pi-Supply PiJuice HAT
tvo@rpi64-deb11:~ $
```
this is on a 64-bit machine, but on a 32-bit armhf system it should be idenntical

ECHEngineering, 2023-11-12:
pi@emonhp:~ $ apt-cache search pijuice
pijuice-base - Basic support for Pi-Supply's PiJuice HAT
pijuice-gui - The desktop applications for the Pi-Supply PiJuice HAT

Yes I was trying to install pijuice-base because I only need the command line interface.  I want to setup the pijuice with ssh connection as that is all I have to these units (because they are already running the emon... software)

tvoverbeek, 2023-11-13:
I now have a pi4 running emonpi with a working PiJuice.
I took the emonSD-10Nov22 (Stable) image from https://docs.openenergymonitor.org/emonsd/download.html, burned a SD then ran an sudo apt uodate, sudo apt -y upgrade. This updated 115 (!) packages.
After a reboot /etc/debian_version showed 11.8 (was 11.5 in the 10Nov22 image).
No pointers whatsoever to Jessie as in your system.
Then sudo apt install pijuice-base worked without problems including installing python3-urwid.
Ofcourse since this is on a pi4 I had to change the symbolic links from the 64 to 32 versions (pijuice_cli32 and pijuiceboot32).
Then everything worked.

ECHEngineering, 2023-11-13:
I wonder if there is something amiss in the cloned sd cards I have here but I have contacted them today and they tell me they are clones and should be the same (clearly they are not).  I have 32 SD cards, I've tried two now in two separate Pi4 with the same result.  Annoyingly this isn't a standard version of the energy monitor as they have customised it for me, so I can't just download and make a new SD card.  
On the Pi I was updating and trying last night, the sudo apt -y upgrade made no updates.  On the fresh one I tried just now, it was similar to yours, many updates.  But still the same error when trying to install.

tvoverbeek, 2023-11-13:
Which error? I did get the pijuice_cli not found error, but that was due to the 32/64 bit problem.
Do you mean the python3-urwid error?

ECHEngineering, 2023-11-13:
Yes, the python3-urwid error.   I'll just ssh and get it on the screen again to copy paste

ECHEngineering, 2023-11-13:
pi@emonhp:~ $ sudo apt install pijuice-base
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 python3-urwid : Depends: python3 (< 3.6) but 3.9.2-3 is to be installed
E: Unable to correct problems, you have held broken packages.
pi@emonhp:~ $

ECHEngineering, 2023-11-13:
I'm wondering if I can clone the SD card of the working unit.  Or whether that would just cause conflicts because I have now set up data feeds from it.... I have asked the question of the emon guys.

ECHEngineering, 2023-11-13:
I've also asked them to try to install pijuice on their version of the software that they cloned for me.... no news yet

ECHEngineering, 2023-11-14:
Okay, credit here to Glyn Hudson of Openenergymonitor...  The issue was that raspbian has just made some changes to the apt repositories to archive the raspbian stretch release that the emonHP is currently running, therefore we need to upgrade to raspbian bullseye before we can install packages.
So, with help from Glyn and extensive help from tvoverbeek here before that, here is my step by step process as it stands.  And it works.   Huge thanks to you both for your expertise in resolving this.  Hopefully the guide below will be of use to others in the future:

First up, edit the sources list:
sudo nano /etc/apt/sources.list

Comment out the first line by adding a # at the beginning e.g 
#deb http://legacy.raspbian.org/raspbian/ stretch main contrib non-free rpi
Then uncomment the second line by removing the # at the beginning
deb http://raspbian.raspberrypi.org/raspbian/ bullseye main contrib non-free rpi
Then run update and upgrade 
sudo apt-get update && sudo apt-get upgrade -y

sudo reboot

We can then install pi juice
sudo apt-get install pijuice-base     (probably works for -gui as well but I only needed to run the -cli version)

manually change the symbolic links      (if using gui, also do this for _gui)
$ sudo rm /usr/bin/pijuice_cli
$ sudo ln -s  /usr/bin/pijuice_cli32 /usr/bin/pijuice_cli
$ sudo ln -s  /usr/bin/pijuice_boot32 /usr/bin/pijuice_boot

Run pijuice to check and edit the configuration:
$ pijuice-cli

...and enjoy.    
May the Pijuice become a part of many more openenergymonitor devices.