PiSupply/PiJuice #1015: Pijuice install does not work. No error except for a "Failed to execute child process /usr/bin/pijuice_gui" (no such file)

Bonzadog, 2023-10-07:
7.10.2023
Raspberry 4, OS: Bullseye updated 7.10.2023

After trying to install Pijuice I get an error message.
history -something is very wrong since PiJuice is entered into the Bullseye "prefences".

sudo apt-get install pijuice-base pijuice-gui
reading package lists...
Building dependency tree...
Reading state information...
pijuice-base is already the newest version (1.8).
pijuice-gui is already the newest version (1.8).
0 upgraded, 0 newly installed, 0 to remove, and 0 not upgraded.

But when in "Preferences" -> Pijuice settings on gets the error message.
"Failed to execute child process /usr/bin/pijuice_gui" (no such file)"

**Here are some folder contents relevant to pijuice**

nt@raspberrypi:~ $ ll /usr/bin/pijuice_gui
lrwxrwxrwx 1 root root 22 Oct  7 16:50 /usr/bin/pijuice_gui -> /usr/bin/pijuice_gui64

ls /usr/bin/pij*
/usr/bin/pijuiceboot    /usr/bin/pijuice_cli32   /usr/bin/pijuice_gui32   /usr/bin/pijuice_sys.py
/usr/bin/pijuiceboot32  /usr/bin/pijuice_cli64   /usr/bin/pijuice_gui64   /usr/bin/pijuice_tray.py
/usr/bin/pijuiceboot64  /usr/bin/pijuice_cli.py  /usr/bin/pijuice_gui.py
/usr/bin/pijuice_cli    /usr/bin/pijuice_gui     /usr/bin/pijuice_log.py

I would appreciate help on this since I need to install a pijuice as soon as possible.
many thanks

tvoverbeek, 2023-10-07:
See issue #1014 and issue #1000 for explanation and fix.
On a Pi4 (and Pi3) the Linux kernel now is 64-bt while userland is still 32-bit.
The install script uses the wrong test to detect when to install 23-bit or 64-bit executables for pijuice_gui and pijuice_cli.

Bonzadog, 2023-10-08:
If this is an install error then when will this get repaired?
Issue 101 an 1000 are not of real help since an OS 32Bit  will be used, this is for me a non-option.
I must have a 64 BIt OS. The latest Bullseye.

tvoverbeek, 2023-10-09:
The current install works fine for Raspberyy Pi OS 64-bit.

The problem is with Raspberry Pi OS 32-bit installs on Pi3 and Pi4.
On Pi3 and Pi4 the 32-bit OS uses a 64-bit Linux kernel with a 32-bit userland.
In the current install procedure (post-install script in the pijuice-base and pijuice-gui packages) the test to detect if  running on a 32-bit or 64-bit installation is looking at the output of `uname -a` which gives the kernel architecture.
On Pi3 and Pi4 this reports `aarch64`. However the pacakge architecture (`dpkg --print-architecture`) reports `armhf` which is the 32-bit arm with hard-float.
Will change the test to use `dpkg --print-architecture`.

Suggest you make a clean install of pijuice-base and pijuice-gui on your 64-bit system.

Bonzadog, 2023-10-09:
> See issue #1014 and issue #1000 for explanation and fix. On a Pi4 (and Pi3) the Linux kernel now is 64-bt while userland is still 32-bit. The install script uses the wrong test to detect when to install 23-bit or 64-bit executables for pijuice_gui and pijuice_cli.

Does not work on the RPi4 Bullseye.

It seems that  the install procedure is wrong  - cannot this be fixed asap? These tips do not work.

Bonzadog, 2023-10-09:
I was wondering if there is an alternative to Pijuice - I cannot wait too long for this project. The tips here did not work.
This GITHUB seems to take some time to answer problems.
a not to happy customer.  I am lucky that the project I have is not too urgent.

tvoverbeek, 2023-10-09:
Answered earlier today. Install works fine 9n my own 64bit system