PiSupply/PiJuice #1016: RTC resets if PiJuice battery charge dies

Wahaj6Ahmad, 2023-10-12:
My project requires Raspberry Pi 3B+, installed remotely, to be powered on every hour and perform a certain task.
The Pi is powered through a PiJuice which is then charged through solar panels.
I need to use the RTC (on Pijuice) to work accurately in order to wakeup the rpi every hour. Problem: In a scenario where there is no sunlight for a long time and PiJuice isn't charged, causing the battery to deplete fully, the RTC will reset. If/When the sunlight is enough to charge the PiJuice, how can I turn on the raspberry pi again?
Note: The Pi is connected to the internet, so the RTC should switch to the correct date and time once the Pi powers on.

tvoverbeek, 2023-10-16:
Look at the "wake-up on charge" option and set an appropriate threshold, e.g. 10%.

Wahaj6Ahmad, 2023-10-19:
The wake-up on charge option does not work once the battery is fully depleted. I simulate this behavior by removing the PiJuice battery. The Pi does not power on automatically after the battery is reinstalled. After manual boot-up, I can see that the wake-up on charge flag goes back to not enabled. The RTC goes to default and the wakeup enabled option becomes unchecked.
Note that I am applying these settings through the PiJuice gui.

In other terms, is there any way of waking up the Pi after the PiJuice battery is removed and reinstalled?

tvoverbeek, 2023-10-19:
Works for me, but you have to turn on the "No battery turn on" option (default is off).
If I shutdown, remove external power and the battery, connceting power to the PiJuice HAT starts the HAT and the RPi powers on.

Wahaj6Ahmad, 2023-10-19:
Okay, so enabling the 'restore' check has solved the problem for me.

Two different questions:
1. What is the purpose of the wakeup-enabled check? (The documentation says for repeated wakeups, the check needs to be enabled after every reboot. However, I did not need to do that. It remains checked despite various reboots.)
2. How can I set the RTC time on PiJuice upon reboot? the Pi is connected to the internet, but it doesn't automatically get set. Also, the time on the Pi is correct, but the RTC on PiJuice gui is incorrect. Any tips?

tvoverbeek, 2023-10-21:
1. In the firmware the wakeup-on-charge flag is cleared at every wake-up. Before firmware 1.6 you had to reenable it after every wakeup. since firmware 1.6 this can bes stored in non-volatile memory, so you do not have to this any more (as you noticed).
2. If the Pi is connected to the interned timesyncd will sync the system clock to internaet time. The system clock setting is then also copied to the RTC.
The RTC is always kept in UTC. So for me in central europe the RTC is 2 hours behind local  time (still  summer time). The RTC time shown in the GUI is in UTC. Also you have to use UTC to set the alarm time.
Does that explain the incorrect time?

Wahaj6Ahmad, 2023-10-24:
Yes, it is clear, thank you!

closed 2023-10-24 by Wahaj6Ahmad